odriverobotics/ODrive #4: Adding USB serial, but need quick overview

nickaknudson, 2017-02-05:
I intend to add USB serial support for ODrive but need a quick overview into how the motor control works. What control modes are currently available? Position control? Velocity control?

I don't see any functions to call into to control the motor. Is all control accomplished by modifying this exported structure?

https://github.com/madcowswe/ODriveFirmware/blob/master/MotorControl/low_level.h#L54

Thanks.

madcowswe, 2017-02-05:
Position, velocity and torque control are implemented.
Gimme an little bit and I'll make some public functions you can call to set the desired values.

nickaknudson, 2017-02-05:
Okay great. I'll get started on the USB interface. Thanks!

madcowswe, 2017-02-05:
Okay done. Check these public functions: https://github.com/madcowswe/ODriveFirmware/blob/master/MotorControl/low_level.h#L75-L78

Call the the corresponding one depending on which mode of control you wish to do.

closed 2017-02-05 by madcowswe