odriverobotics/ODrive #17: Clean up and robustify USB communication

madcowswe, 2017-03-27:
It should be able to be configured and commissioned over USB, and it should be able to take runtime commands from USB.

If that means that we have a G-Code parser on board, or if we hook it up through some python on the PC that then speaks some binary protocol is yet to be determined. Feel free to add suggestions.

dirk-attraktor, 2017-04-03:
I added a simple get/set usb interface + a python code generator to create an api + a javascript webinterface with live plotting. 

I also added some code for monitoring/live plotting: my idea was this: because there is alway a compromise between "how many values" and "how often" , i added  monitoring_slots, 20 for now. The usb-api exposes a function like "monitor value xy on slot 1". If you then call "o(utput) 2" you get the first 2 monitored values. So a client can configure it to have important values on the first few slots, and read these with kHz, and less important values later in the array, so maybe every once a second it can call "o 20" to get all monitored values. The ODrive.py code makes use of these functions and automatically assigns and reads monitoring slots if you choose to monitor some value with some frequency. 

Pull request: https://github.com/madcowswe/ODriveFirmware/pull/19

alexisdal, 2017-04-11:
iwth master commit b0e57908ce5612043f9de9885dc72b257ac80cb9
i managed to get USB interface unresponsive with motor not moving anymore by using the following script

```
#!/bin/bash
while(true); do
   num_points="9999"
   echo -en "p 0 $num_points 0 0\n" > /dev/ttyACM0
   sleep 3.0
   echo -en "p 0 -$num_points 0 0\n" > /dev/ttyACM0
   sleep 3.0
done
```

(basically going back and forth)

i get the usb interface to stop responding after about 1 or 2 minutes

alexisdal, 2017-04-11:
i think it is not relevant but i can post specific motor params in low_level.c along with engine and encoder specs if that help. plz advise

madcowswe, 2017-08-24:
Okay, so for right now, here's the plan:

- We wanna keep the ASCII style comms for now.
- Main thing to improve is presenting an API that has symbols for the parameters instead of numbers
- We make all parameters use getters/setters (this is important for setting computed parameters, like control bandwith)
- We store a table of structs containing either:
  - {"name_string", TYPE_AS_ENUM, address_of_getter (or NULL), address_of_setter (or NULL)}
  - {"name_string", CONTAINER_SPECIAL_ENUM, subtree_ptr, subtree_len}
  - The last two elements would be a union.

alexisdal, 2017-08-31:
i just reproduced the issue using 

* odrive board v3.2
* turniny aerodrive 5065 236
* a simple lasercut hardward jig to hold the motor and the encoder.
* a rohde&shwarz lab power supply (settings on 24v / max 4a)
* a R47 resistor connected on AUX (since i am using a power supply instead of a battery, my understanding was that i should use a "big" resistor connected on AUX)
* lastest master code (06047cb79139837de582afbf32202c237e1b5627) updated by Oskar 6 days ago. no modification at all.
* the same code snippet as above (basically sending "p 0 9999 0 0\n" and "p 0 -9999 0 0\n" back and forth every 3 seconds)

After about 20~30 seconds, the motor stops moving and usb interface no longer responds. the only way to work with the board is tostop sending request, turn off and on power supply, wait for the init sequence to be over. 

I don't care much about the API interface as long as it is simple and stable. what madcowswe suggests above is very fine for me. I have not tried the step/dir hardware interface yet.

Basically the symptom is the USB interface is not usable.

Should I use another brand? 
Should I use the step/dir interface instead of USB ?
Does anyone reproduce the issue?

madcowswe, 2017-09-24:
Improved struct contents:
{"name_string", TYPE_AS_ENUM, address_of_variable (or general void*), NULL (or general callback)}

madcowswe, 2017-11-08:
#52 addresses this.

closed 2017-12-02 by madcowswe