odriverobotics/ODrive #20: Automated detection of common mistakes

madcowswe, 2017-04-06:


Hello1024, 2017-04-06:
You should also have a self test mode, probably run on every startup.

SimonK tests each mosfet for being open or shorted, tests each ADC channel, and tests the continuity of motor windings.

We could also test motor coil inductance (detects burned motors), motor coil resistance (detects poor wiring).

We can also detect power supply internal resistance (detects unsuitable supply) and power supply voltage being out of range.

If the brake mode is used, measure the resistor value and check the mosfets.

We can also check for motor encoders being connected backwards, or only one of two quadrature inputs being connected.

I *think* all that stuff can be measured/approximated with no hardware modifications and no user interaction.

madcowswe, 2017-04-06:
Yes, I agree on all counts.