odriverobotics/ODrive #22: Use separate file for config/parameter values

madcowswe, 2017-04-08:
Currently the configuration is done with a bunch of values at the top of motor_control.c. This is quite messy, especially if you want to switch between different configs.
The best way would be to have config files.
Maybe is a list of CONFIG_XXXX defines, and get #included in low_level.c?

dirk-attraktor, 2017-04-09:
Jep, config file to include sounds good, i will also add some save/load config stuff to the python api, so the configuration can also be loaded without recompiling if the odrive is controlled by some higher level system.

Wetmelon, 2017-08-22:
This has happened as part of the refactor.

madcowswe, 2017-08-24:
I think the best way to go is to split the structs into:
`Motor_t, MotorConfig_t` -- `Sensorless_t, SensorlessConfig_t`, etc.
and then present just the config structs to the user.
This way we get type safety, and can copy entire configs around easier.

beak90, 2018-01-11:
Until this is finished, what's the best way for me to keep my settings out of the public codebase, and to keep my settings intact when pulling updates from the master? Seems like there are ways to keep whole files separated from master, but I don't want to prevent all of low_level.c from being updated.

closed 2020-07-31 by Wetmelon