odriverobotics/ODrive #29: Refactor codebase to C++

madcowswe, 2017-09-23:
Progressively move the code over to C++. At the same time compartmentalise and organise, make more modular.

madcowswe, 2017-10-09:
'''C++
//Outside axis:
    //command handler
    //callback dispatch

class Axis {
public:
    //thread/os/system management
        //timing log
        //thread id
        //etc.
    //state machine
        //control mode
        //control_en/calib_ok
        //error state
    //motor
        //current controller
            //contains rotor phase logic
        //motor level calibration routines
        //low_level (implementation specifics)
            //DRV driver
            //adc callback handling
            //pwm queueing
    //rotor estimator
        //kick out rotor phase logic
    //pos/vel controller
    //step/dir handler
'''

samuelsadok, 2018-03-10:
addressed in #136

closed 2018-06-06 by madcowswe