odriverobotics/ODrive #32: Plotting tools

madcowswe, 2017-10-05:
Plotting of states, measurements and control effort in Python host.

- [ ] Liveplotting of selected exposed variables over USB
- [ ] Embedded variable oscilloscope: A oscilloscope-like buffer in firmware that can trigger on an event and hence log at the full sample rate of the control loop into a buffer: then transfer the buffer to the host.

Depends: #30

madcowswe, 2017-10-29:
Liveplotting implementation suggestion:
The communication protocol allows "subscription" to a variable, with a specified trigger. One of the triggers is that the control loop has finished an execution. This is communicated to the communication thread from the control thread via a semaphore at the end of control loop.
The communication thread sends all data subscribed to on this trigger when the semaphore is triggered.

Wetmelon, 2020-11-10:
You added the oscilloscope functionality, right?  Also, is this still relevant?

closed 2021-09-13 by Wetmelon