odriverobotics/ODrive #34: Excitation signals

madcowswe, 2017-10-05:
We should come up with some set of excitation/stimulus signals that the firmware can do to generate currents and movements that will help reveal model parameters.

Just some super basic pulse current stimulus at close to the motor’s current rating (to get strong acceleration/mass visibility) that iteratively gets longer and longer in duration until we have also reached some threshold speed/displacement. Once that has identified the basic response, we can tune the basic feedback gains, and hence do closed-loop stimilus. For these, a set of a few different frequency and amplitude sinusoids could be appropriate. Maybe throw in some steps?

Wetmelon, 2020-11-10:
Added "Input Mode Tuning" with `A*sin(2*pi*f + phi)` forwarding to #519

Wetmelon, 2021-09-13:
Adopted in `INPUT_MODE_TUNING`

closed 2021-09-13 by Wetmelon