odriverobotics/ODrive #38: [Feature Request] - Function to Set Encoder Register

youngiif, 2017-10-10:
I need a function to write to the encoder position register.

At least two threads on discourse discuss this desire:
https://discourse.odriverobotics.com/t/motor-rotor-encoder-state-read-only/269
https://discourse.odriverobotics.com/t/startup-against-gravity-with-encoder-with-index-pulse/251

I don't mind trying to create this feature and submit a pull request, I could just use a hand in getting started as I have no experience writing firmware for ARM chips.  I hopefully can trace/follow existing code enough to figure it out though.

If anyone can offer a bit of guidance, I am assuming I would need to create a function that calls DRV8301_writeSpi()?

madcowswe, 2017-10-11:
Thank you for volunteering to have a go at creating this.
So basically the 32 bit variable motor->encoder.encoder_state tracks the 16 bit count register in the STM32's timer pheriperal, located at motor->motor_timer.Instance->CNT.
The tracking is done [here](https://github.com/madcowswe/ODriveFirmware/blob/devel/MotorControl/low_level.c#L1107-L1108).

If you wish to update the encoder state, you must make sure to update the underlying register as well as the encoder_state.

madcowswe, 2017-11-26:
#75 implements this.
@youngiif can you please verify that this is able to do what you need?

closed 2017-12-02 by madcowswe