odriverobotics/ODrive #53: Anticogging improvments

madcowswe, 2017-11-06:
- [ ] Use setpoint instead of current position to lookup table, in order to keep the compensation on the feed-forward path only
- [ ] Selected harmonics smoothing
  - Upload calibration map
  - FFT - select harmonics - IFFT in python
  - Download smooothed map

Wetmelon, 2018-01-06:
Thinking about this... would it be best to use a position prediction based on p & v, then take say... (prediction + current pos) / 2 as the lookup, or similar?

madcowswe, 2018-08-17:
A more concrete plan for the 2nd part:
1. modify the firmware to scan into a small circular buffer with a "last written idx"
2. get python to poll it and read the buffer to capture the data
3. run and plot my analysis script as usual, make sure that still works
4. said script generates the smoothed version of the cogmap: resample this into a smaller cogbuffer, size TBD, something like 128 samples maybe 256.
5. Write this resampled cogmap back to ODrive.
6. modify the anticogging path that's called in every control cycle to use this smaller map and to simply linearly interpolate the nearest samples
7. figure out a way to store the cogmap as a config member as to make it stay in persistent memory

Wetmelon, 2018-09-08:
If used in conjuction with trajPlan, anticogging is now placed in the feedforward path.

madcowswe, 2020-08-31:
We plan to use a continuously rolling "integrator" with LPF broadcast of integrand.

closed 2020-08-31 by madcowswe