odriverobotics/ODrive #55: Separate printf from other communication.

madcowswe, 2017-11-08:
Printf is usueful for debugging and casual status printing. However, now the transport used for printf is shared with the primary communication protocols.
We can, instead, dedicate Endpoint0 of USB to be printf, and use Endpoint1 for opreational communication.

Optionally, we can also have a mode where printf goes onto UART instead of USB, if desired.

samuelsadok, 2018-05-14:
By now on current devel you can use `CONFIG_USB_PROTOCOL=stdout` or `CONFIG_UART_PROTOCOL=stdout`. Given that we mostly debug using cortex-debug and the native protocol I believe that we have most use cases covered.

madcowswe, 2018-06-15:
Closed with release of fw-v0.4.0

closed 2018-06-15 by madcowswe