odriverobotics/ODrive #64: Support encoder index pulse

madcowswe, 2017-11-19:


Wetmelon, 2017-12-14:
This is working on my end, just need to figure out how to fix my branch issue (right now, it will try to include USB Descriptor reporting PR too because I put it on devel like a dingus)

fluture, 2017-12-31:
Can the index pulse be configurable?
I am using [AS5304](https://www.mouser.com/ProductDetail/985-AS5304AB) with these [magnetic rings](http://ams.com/eng/Products/Magnetic-Position-Sensors/Magnets/AS5000-MR20-44)
These have 44 index pulses per revolution (it depends on the number of poles)
While  [AS5047](https://www.mouser.com/productdetail/ams/as5047p-ts_ek_ab) has only one index pulse per revolution while using the ABI interface

madcowswe, 2017-12-31:
@fluture You cannot have fewer index pulses than the number of pole pairs, because the encoder ring aliases. I.e. there is no way for the encoder to know which cycle you are in.

fluture, 2017-12-31:
I don't want fewer, I was just thinking how or if it would work with magnetic rings which have more than one index pulse per revolution, I understand that the most common way is to have one per revolution (360 mechanical degrees) like the AS5047

madcowswe, 2017-12-31:
I see. If you get the same number of pole pairs in the ring as there are in the motor, you can use the index pulse to restore the encoder calibration, just like a 1-index-pulse encoder.
However, you cannot restore the anti-cogging calibration.

Wetmelon, 2018-01-31:
Closed by #106

closed 2018-02-04 by madcowswe