odriverobotics/ODrive #70: Hall sensor feedback instead of encoder

madcowswe, 2017-11-19:
Some applications do not require very precise positioning, but there are readily available motors with hall sensors. Example is hub motors for use with mobile robots.

wernerdaehn, 2018-06-29:
I am a VESC user and just ordered an ODrive with encoder. Of course an encoder will give you smoother low speed movement and more positional precision.
What would be important to me however is that the position holding (speed=0) works well. I do not care if the absolute position is off by an inch as long as it stops or does move very, very little.
Frankly, I am not sure this is technically even possible without tuning the controller to such an extreme, that it cannot deal with different loads anymore. 
However wanted to raise the awareness.

https://www.youtube.com/channel/UCTLw_Boy24qAhuygCyIOzyw

madcowswe, 2018-07-01:
There is now a preview for the hoverboard motor config writeup [here](https://github.com/madcowswe/ODrive/blob/devel/docs/hoverboard.md).

madcowswe, 2018-08-10:
Implemented in fw-v0.4.2

closed 2018-08-10 by madcowswe