odriverobotics/ODrive #72: USB bootloader

madcowswe, 2017-11-21:
Some known implementations, in no particular order. Some might suck.

* Google for "stm usb bootloader", maybe there is something very easy to install.
* webdfu / webusb
* maple mini bootloader (used by VESC via ChibiOS, [link](https://github.com/vedderb/bldc/blob/master/ChibiOS_3.0.2/os/common/ports/ARMCMx/compilers/GCC/ld/STM32F103xE_maplemini_bootloader.ld)
* meseta's implementation: https://github.com/meseta

madcowswe, 2017-11-21:
Notes by crinq:
in the long run you should use a custom post bootloader that checks fw crc and board revision
error -> jump to rom bootloader
ok -> jump to main fw
board revision is hardcoded in the post bootloader and in the fw info block
crc only: https://github.com/rene-dev/stmbl/tree/master/bootloader

samuelsadok, 2017-12-04:
The actual jump to the system memory (where the bootloader lives) happens here:
https://github.com/rene-dev/stmbl/blob/master/bootloader/src/main.c#L65-L72
Apparently it uses the first flash sectors to CRCheck the rest and invoke the STM bootloader if necessary (however the code there is GPLv3 so we can't directly use it).

More low level documentation: [AN2606](http://www.st.com/content/ccc/resource/technical/document/application_note/b9/9b/16/3a/12/1e/40/0c/CD00167594.pdf/files/CD00167594.pdf)

(Note: from a security standpoint the STM bootloader seems kinda hopeless)

madcowswe, 2018-03-14:
Adressed by #82

closed 2018-03-14 by madcowswe