odriverobotics/ODrive #73: Compile a datasheet for each release.

madcowswe, 2017-11-21:
Some people are wondering what the capabilities of the ODrive are: Specifically things like supported commands and continous current capacity, etc.
We should compile a datasheet for each release, condensing the capabilities of the drive at that point. While this may technically be redundant information from the Changelogs, etc, it will help many who see the project for the first time get a good idea of what it can do.

madcowswe, 2017-11-22:
Suggestion: Keep an active LaTeX document for this datasheet, compile and upload .pdf with each release.
Sample datasheet: http://downloads.ionmc.com/docs/roboclaw_datasheet_2x15A.pdf

h0dgep0dge, 2019-05-13:
Ironically, I've been searching for this exact information, and instead I find someone saying it needs to be easier to find lmao. Anyone have any hints on where I could look for this sort of more concrete information?

Wetmelon, 2021-09-13:
Currently available on docs.odriverobotics.com, although it's not super easy to find the "basics"