odriverobotics/ODrive #80: Trajectory generation/following

madcowswe, 2017-12-02:
Various kinds of trajectories could be suitable. To start with, we should implement a sinusoidal velocity profile, point-to-point (stopped-to-stopped) generator and follower.

* [Reference paper](http://www.aspe.net/publications/Spring_2001/01Sp%20Extended%20Abstracts/Arevalo.PDF)

madcowswe, 2018-08-19:
We decided to start with a simple trapezoidal trajectory, initial work in #210.

wernerdaehn, 2018-10-13:
I am starting to integrate the odrive into my cablecam controller. Velocity ramp support is one of the areas I need. One option would be to use the odrv0.axis0.controller.pos_setpoint command at a fixed interval and hence simulate the ramps. No problem with that. But I am trying to understand if the Trajectory approach is meant for that.
What I expect is to set the velocity ramp, e.g. 50counts/sec/sec (change the speed by 50counts/sec every second) and then I enter a goto command and the motor will accelerate slowly up to the top speed and decelerate to reach the target point.

Are we talking about the same thing here? Why is a trajectory planning needed in the python code? Why would you want a non-trapezoid speed trajectory?

Wetmelon, 2018-10-24:
@wernerdaehn Your "velocity ramp" is actually already available in the code.  Use <odrv>.<axis>.controller.move_to_pos().  Please see the discourse thread here

https://discourse.odriverobotics.com/t/trapezoidal-trajectory-planner/1104

wernerdaehn, 2018-10-24:
Perfect. Thanks.

Wetmelon, 2019-09-08:
This has been done for a while (for initial trajectory).  We will open a new ticket for further trajectories

closed 2019-09-08 by Wetmelon