odriverobotics/ODrive #94: Move Python library to its own repo

beak90, 2018-01-10:
It seems like it would be cleaner to keep all the tools in their own repositories. So the Python library would have its own, just like the Arduino library does now. Also, the Python library can be made pip-installable, for easier dependency installation and easier usage.

madcowswe, 2018-01-10:
The reason we have it in the same repository is to allow the changes to the protocol to be coherent. Do you have any suggestions on how to best achieve that if we separate the repos?

beak90, 2018-01-10:
The protocol is kept accurate in all libraries manually, correct? So putting it in a different repository wouldn't be any easier or harder from that perspective.
The other solution would be to put the Arduino library into this repo. The main reasoning is that as more libraries are developed it could become a mess. I think some other client libraries would be useful (maybe Nodejs, Arduino but with CAN, MATLAB, etc).

madcowswe, 2018-01-10:
Yes, updating the relevant libraries and pushing the changes is all manual, so I agree it makes no difference. I think the main difference is when pulling in the update, it can get incoherent very easily. From my experience with both ways of doing it, and at least for a developer's perspective, the monolithic repository was easier to work with.

Yes Indeed I would prefer to also have the Arduino library included. The reason I didn't is because the arduino library importer seems to want a github repository with a specific structure. I haven't registered the library with the arduino library manager yet, and actually I don't really know the requirements that well: if we can make it point to a sub-folder of a larger repo that would be great.

beak90, 2018-01-11:
Ah ya I see that here: https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ
It specifically says you can't point to a subfolder in a larger repo. The library.properties file must be in the root of the repo. And it will assume the whole repo is the library.

The process for PyPi is more flexible. You can host the tarball wherever you want. But if the whole github repo is just the python library, then the tarball is auto-generated:

> Github creates tarballs for download at https://github.com/{username}/{module_name}/archive/{tag}.tar.gz

http://peterdowns.com/posts/first-time-with-pypi.html

madcowswe, 2018-01-15:
As per discussion, we will leave the Python and Firmware together.

closed 2018-01-15 by madcowswe