odriverobotics/ODrive #97: USB Protocol doesn't work on Mac

beak90, 2018-01-12:
I'm opening a separate issue for this protocol. This is for the USB_PROTOCOL_NATIVE protocol.

This protocol doesn't work with Mac. The issue seems to be because the operating system matches the device with a built-in driver. PyUSB/libusb can't connect to it because it is already connected to a driver. MacOS doesn't seem to let libusb disconnect the kernel driver from it, the way it can in Linux.

I've tried unloading both the official FTDI driver (/Library/Extensions/FTDIUSBSerialDriver.kext) and the built-in Mac FTDI driver (/System/Library/Extensions/AppleUSBFTDI.kext). This doesn't fix it. I did notice a kext named com.apple.driver.usb.cdc, but it wouldn't let me unload it ("in use or retained").

Unloading common USB drivers seems like a poor fix for this. The real fix is to make, sign, and distribute a "codeless kext", which doesn't do anything at all, but that matches the device by id, name, etc. That prevents the system driver from connecting.

Some of these methods are mentioned here: https://github.com/libusb/libusb/wiki/FAQ#How_can_I_run_libusb_applications_under_Mac_OS_X_if_there_is_already_a_kernel_extension_installed_for_the_device

Also, I don't know if these solutions will work at all. There could be some other issue that will prevent this protocol from working with Mac.

beak90, 2018-01-12:
Here's some info from the Discord about the way this protocol was devised, from Oskar:

> ok, so here's how it goes:
The odrive gives the PC a USB descriptor that says "I'm a CDC device, here are my standard capabilities"
the PC is like cool, we know what to do here, lets use the OS provided CDC driver: the ODrive shows up as a serial port
as we send data as a serial port, the PC packetises it up and sends it over the CDC spec to the ODrive
STM doesn't actually assemble it back into a character stream for you, they actually only help making the descriptor. The callbacks just give you a raw data buffer of what was sent in that specific packet.
so we were like, huh whatever, let's just send data with packets then, we'll get more control anyway
so we (on Windows and Linux) tell the PC: No, we actually don't want the OS's CDC driver, we'll do this ourselves with libusb/pyusb
so basically we use STM's CDC setup to get everuthing up and running, then we go "jk, let's do it raw packets"
There are two cleaner alternatives of course:

>1. we never claim to be a CDC device, and so no PC would try to connect with a serial port driver. Then we use libusb only. The main thing stopping us from that is we have working CDC boilerplate from ST, and we don't wanna spend the time to go change it to be a valid USB device without claiming to be a CDC device.
>2. We don't send raw, we use a stream based protocol as the main protocol on all platforms. The ODrive is a serial port on all platforms. This could work, but there are some nice parts to sending stuff as well-known packets

- Oskar, Jan 11 #development

beak90, 2018-01-12:
The good news is that it does work with a Virtual Machine. I'm using Parallels, since that is what is supported best by Macs, but I imagine that VirtualBox or VMWare would also work.

I used Ubuntu 16. I followed the setup instructions, but I did have to use sudo python3 to get it to work. Without sudo it said "USB device access denied".

beak90, 2018-01-12:
Some updates.
I was able to find where the USB info is, in MacOS. Here's a screenshot of IORegistryExplorer. You might need to download it separately, but it comes from Apple.

[image]

This doesn't clarify which system driver is attached to it, though. I did some more testing with kextstat, and I was able to find that the /dev/tty.usbmodem entry disappears when I unload com.apple.driver.usb.cdc.acm
The python lib still doesn't find it though. I also tried unloading com.apple.driver.usb.cdc, but that also didn't help. Unloading this kext is possible only after unloading com.apple.driver.usb.cdc.acm. I saw somewhere that you need to restart the computer in order for the changes to take effect, but this doesn't work. It just reloads the kexts after reboot.

Now that I have all the device id, etc, info for the ODrive, I will try to put together a codeless kext. If that works, then that would be a super easy installation.

samuelsadok, 2018-04-27:
Thanks for looking into this.

There should actually another way to keep the kernel from attaching a driver: setting the device class to a value for which the kernel has no driver.
To facilitate this, I made the ODrive act as a composite device with two interface groups. One group (interfaces 0 and 1) is still a CDC device, so the kernel should still attach its driver there (and you should still see /dev/tty.usbmodem). The second group (interface 2) however has a vendor specific class ID, so the kernel should leave it untouched.

I don't have macOS (anymore) to test though. Would you mind trying out this commit:
https://github.com/madcowswe/ODrive/commit/b95d3539e07efa24ca3cbd91324bd6ef1eda52b2 ?
Basically you'd need to `make flash` and then hopefully `tools/odrivetool` will find the device. If not, you can run `tools/odrivetool.py -v` to get debug info.

madcowswe, 2018-06-06:
This is hopefully adressed in #166.

closed 2018-06-06 by madcowswe