odriverobotics/ODrive #107: Let GPIO5 be used as enable line when using step/dir

madcowswe, 2018-02-05:


gottmoz, 2018-02-06:
Please make the feature so that it is possible to choose enable pin attive High/low/Charge pump.
That way all controller can use the feature.

madcowswe, 2018-02-06:
@gottmoz I understand the active high/low part, but I don't understand what "charge pump" is in terms of signaling polarity?

gottmoz, 2018-02-06:
Mach 3 cnc and other cnc software use chargepump wich means that they expect the signal to pulse high/low to make sure the software is running ok, if pulsetrain stops the servo drive should be disabled.

Skickat från min iPhone

Wetmelon, 2018-02-07:
How often do they pulse the signal?  Is it on every step?

gottmoz, 2018-02-08:
The signal should bre 12.5kHz all the time when drive is enabled...

Skickat från min iPhone