odriverobotics/ODrive #119: Add range checks to communication protocols

madcowswe, 2018-02-24:
Make the defaults -inf to inf (or minval/maxval for integers). Make sure NaN etc gets rejected.

fluture, 2018-02-26:
Sending a negative value for vel_lim makes the motors turn, it should report an error code and/or convert to absolute value and definitely not move

Wetmelon, 2020-11-10:
@samuelsadok is there an easy way to do this outside of creating a getter/setter for each function?

samuelsadok, 2020-11-11:
We could add this functionality to the interface_generator tooling. Conceptually I would think of the min/max values as a template argument to the `fibre.Property` interface which are given in `odrive-interface.yaml`, though the actual C++ implementation might not use templates (to save on code size).