odriverobotics/ODrive #135: Count step/dir in encoder TIM hardware

madcowswe, 2018-03-09:
Crinq says:
INput 1/2 as per https://github.com/rene-dev/stmbl/blob/master/src/comps/enc_cmd.c#L126
Check reference manual for input capture and Encoder mode.

madcowswe, 2020-11-24:
Crinq [on discord](https://discordapp.com/channels/369667319280173067/369678848972816385/780826205443063819):
```
CR1:
  CEN = 1
CCER:
  CC1E = 1
  CC1P = 1
  CC2E = 1
SMCR: 
  SMS = 3
CCMR1:
  CC1S = 2
  CC2S = 1
ARR = XYZ
```

madcowswe, 2020-11-26:
Some more info here:
https://stackoverflow.com/questions/32947972/stm32-how-to-make-pulse-count-up-down-with-timer

Critical note:
> The TIM would still act as a step and direction counter BUT would count in different directions for the same input level on the direction pin between bootups.

We should validate both the stmbl variant, and the stackoverflow variant.

madcowswe, 2020-11-26:
Crinq says this code https://github.com/rene-dev/stmbl/blob/master/src/comps/enc_cmd.c#L140-L161
to ignore the comments they are out of date.
Also:
> step signal is tim2_ch1, dir is tim2_ch2

Wetmelon, 2021-09-13:
Wontfix.  We improved the overhead associated with software interrupts and incrementing the step/dir, to now support up to 250kHz on a channel.

closed 2021-09-13 by Wetmelon