odriverobotics/ODrive #157: Initial Endpoint documenation

sevenbitbyte, 2018-04-14:
Please document the varios endpoints and their expected behaviors. The existing [protocol documentation is excellent](https://github.com/madcowswe/ODrive/blob/master/Firmware/protocol.md) but doesn't cover what endpoints to expect or how they work.

Most endpoints appear to be implemented [here in the firmware](https://github.com/madcowswe/ODrive/blob/master/Firmware/MotorControl/commands.cpp#L120)

madcowswe, 2018-04-15:
Yes we should add some documentation indeed, and we will be working on that [here](http://docs.odriverobotics.com/commands). This pertains to the firmware version 4.0 and later, which is currently under construction, preview [here](https://github.com/madcowswe/ODrive/tree/sam_refactoring). One thing in particular to note with this refactoring is that the endpoints are now declared within each object, example [here](https://github.com/madcowswe/ODrive/blob/sam_refactoring/Firmware/MotorControl/axis.hpp#L167)

madcowswe, 2020-08-29:
With v0.5.1 endpoints can now be directly documented in the `odrive-interface.yaml`. Some initial documentation has been done.

closed 2020-08-29 by madcowswe