odriverobotics/ODrive #171: Standalone application thread

madcowswe, 2018-06-05:
Some application logic is suited for running directly on the ODrive. We should make an application thread with an example demo program, and add some example IO parameters.

madcowswe, 2018-12-20:
Notes on App thread:
You can add it in the bottom of main.cpp. Can you make the official skeleton version that just has a dummy stub for people to add their application code in? Like this:
```
run_application() {
// Put your application code here

// Currently this does nothing
osDelay(100);
}
```
you can start it at the bottom of odrive_main (in main.cpp)
```
osThreadDef(application_thread, run_application, osPriorityLow, 0, 4096);
app_thread_id= osThreadCreate(osThread(application_thread), NULL);
```

Add to the top area of main.cpp
```
osThreadId app_thead_id;
```
and in vApplicationIdleHook add the appropriate entry using app_thead_id. You will probably need to extend the system_stats_ struct.

Wetmelon, 2019-03-25:
Did #318 take care of this?

Wetmelon, 2020-11-10:
#472 makes large changes.  @samuelsadok we should consider adding the documentation as requested above

samuelsadok, 2020-11-11:
I started a simple list here: https://github.com/madcowswe/ODrive/blob/control-loop-refactor/docs/resources.md

closed 2021-09-13 by Wetmelon