odriverobotics/ODrive #183: odrive.save_configuration() fails when called 3 or more times.

Nate711, 2018-06-23:
The save_configuration() function of the python odrive library fails if you call it 3 or more times.

It seems that in these cases, only the first change is saved permanently. The code below reproduces the problem by changing the brake_resistance 3 times and saving after each change. Even though the last change sets the brake_resistance to 2.0, the odrive will say the brake_resistance is still 0.0.

CODE:

import odrive
from fibre import Logger, Event
from odrive.utils import OperationAbortedException
from fibre.protocol import ChannelBrokenException

def get_odrive(shutdown_token):
    print('Looking for ODrive...',end='')
    odrv = odrive.find_any(search_cancellation_token=app_shutdown_token, channel_termination_token=app_shutdown_token)
    print('Found.')
    return odrv

def reboot_odrive(odrv):
    """
    Reboot odrive
    """
    try:
        odrv.reboot()
    except ChannelBrokenException:
        print('Lost connection because of reboot...')

app_shutdown_token = Event()
try:
    odrv0 = get_odrive(app_shutdown_token)
    print('Erasing configuration...',end='')
    odrv0.erase_configuration()
    print('Done.')
    reboot_odrive(odrv0)
    odrv0 = get_odrive(app_shutdown_token)

    print('*** Calling save_configuration() 3 times...',end='')
    odrv0.config.brake_resistance = 0.0
    odrv0.save_configuration()
    odrv0.config.brake_resistance = 1.0
    odrv0.save_configuration()
    odrv0.config.brake_resistance = 2.0
    odrv0.save_configuration()
    odrv0.config.brake_resistance = 3.0
    odrv0.save_configuration()
    print('Done.')

    reboot_odrive(odrv0)

    odrv0 = get_odrive(app_shutdown_token)
    print('Brake resistance (should be 2.0): ',end='')
    print(odrv0.config.brake_resistance)

    # init_odrive(odrv0)
except OperationAbortedException:
    logger.info("Operation aborted.")
finally:
    app_shutdown_token.set()

adammunich, 2019-06-12:
Commenting these if/else lines seems to fix it in nvm.c

```
    // invalidate the other sector
    //if (read_sector->index < read_sector->n_data)
        status = set_allocation_state(read_sector, read_sector->index, 1, INVALID);
    //else
        status = erase(read_sector);
```

xoxota99, 2019-07-25:
Any ETA for a fix?

samuelsadok, 2019-08-13:
> Any ETA for a fix?

0 days (pending merge)

JB-VA, 2019-12-03:
Does this also apply to odrivetool? 
I'm finding my `vel_gain = 0.0` after a reboot even though I've saved before.

samuelsadok, 2019-12-04:
Yes it applies to any method by which you command the ODrive. To make sure what you're seeing is this bug (and as a workaround) try to save only once and then reboot. In that case the setting should persist.

JB-VA, 2019-12-04:
As a new user to odrive I did not know about this and I think I have wasted a lot of time trying to get my drive working.

It's been well over a year since the bug was originally opened. The pull request is just sitting there.. Are we going to see any action taken?

madcowswe, 2019-12-04:
Apologies, it seems I forgot to merge this PR (#371). It is now merged to `devel`, and should make it out on the next release.

JB-VA, 2019-12-04:
Awesome. Thanks guys for the prompt response.

JB-VA, 2020-05-15:
Hi I'm finding this still reoccurs with `fw-v0.4.12` are you able to confirm?

samuelsadok, 2020-05-15:
Yeah sorry v0.4.12 was just a hotfix release for a trajectory bug. We could have pulled this in too but forgot about it.

closed 2020-08-29 by madcowswe