odriverobotics/ODrive #195: Odrivetool not working! failure to connect to board.

Shreeyak, 2018-07-14:
I just updated to v0.4.1, and the board will not connect to the pc using the odrivetool. The USB can be detected by the system, but the tool hangs while trying to connect to the board. v0.4.0 also does not work. In this state, I find the entire board to be completely unusable.

Please find more details in this thread: https://discourse.odriverobotics.com/t/odrive-doesnt-show-up-in-odrivetool/808/17?u=shreeyak

Shreeyak, 2018-07-14:
Found some more details:
I ran odrive tool so:
```
 $ sudo -E env "PATH=$PATH" odrivetool -v
ODrive control utility v0.4.1
Waiting for ODrive...
USB discover loop
ConfigurationValue 1
	InterfaceNumber 0,0
		EndpointAddress 130
	InterfaceNumber 1,0
		EndpointAddress 1
		EndpointAddress 129
	InterfaceNumber 2,0
		EndpointAddress 3
		EndpointAddress 131

Please connect your ODrive.
You can also type help() or quit().

Kernel Driver was not attached
EndpointAddress for writing 3
EndpointAddress for reading 131
Connecting to device on USB device bus 1 device 10
In [1]: 

```

This was the output on dmesg
```
$ dmesg
...
[17231.616732] usb 1-2: new full-speed USB device number 10 using xhci_hcd
[17231.762320] usb 1-2: New USB device found, idVendor=1209, idProduct=0d32
[17231.762321] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[17231.762322] usb 1-2: Product: ODrive 3.4 CDC Interface
[17231.762323] usb 1-2: Manufacturer: ODrive Robotics
[17231.762323] usb 1-2: SerialNumber: 306539843235
[17231.762861] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[17232.776925] usb 1-2: reset full-speed USB device number 10 using xhci_hcd
[17232.917842] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[17236.332965] usb 1-2: reset full-speed USB device number 10 using xhci_hcd
[17236.474023] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[17246.093150] usb 1-2: reset full-speed USB device number 10 using xhci_hcd
[17246.242083] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[17351.473955] audit: type=1400 audit(1531543251.759:88): apparmor="DENIED" operation="open" profile="snap.gitkraken.gitkraken" name="/etc/" pid=16408 comm="sh" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[17353.400471] audit: type=1400 audit(1531543253.685:89): apparmor="DENIED" operation="connect" profile="snap.gitkraken.gitkraken" name="/run/user/1000/keyring/ssh" pid=27530 comm="gitkraken" requested_mask="wr" denied_mask="wr" fsuid=1000 ouid=1000
[17410.809309] audit: type=1400 audit(1531543311.093:90): apparmor="DENIED" operation="open" profile="snap.gitkraken.gitkraken" name="/etc/" pid=16507 comm="sh" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[17412.633144] audit: type=1400 audit(1531543312.917:91): apparmor="DENIED" operation="connect" profile="snap.gitkraken.gitkraken" name="/run/user/1000/keyring/ssh" pid=27530 comm="gitkraken" requested_mask="wr" denied_mask="wr" fsuid=1000 ouid=1000
[17470.823418] audit: type=1400 audit(1531543371.106:92): apparmor="DENIED" operation="open" profile="snap.gitkraken.gitkraken" name="/etc/" pid=16604 comm="sh" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[17472.656061] audit: type=1400 audit(1531543372.939:93): apparmor="DENIED" operation="connect" profile="snap.gitkraken.gitkraken" name="/run/user/1000/keyring/ssh" pid=27530 comm="gitkraken" requested_mask="wr" denied_mask="wr" fsuid=1000 ouid=1000
[17530.829600] audit: type=1400 audit(1531543431.111:94): apparmor="DENIED" operation="open" profile="snap.gitkraken.gitkraken" name="/etc/" pid=17037 comm="sh" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[17532.683612] audit: type=1400 audit(1531543432.965:95): apparmor="DENIED" operation="connect" profile="snap.gitkraken.gitkraken" name="/run/user/1000/keyring/ssh" pid=27530 comm="gitkraken" requested_mask="wr" denied_mask="wr" fsuid=1000 ouid=1000
[17558.630120] usb 1-2: reset full-speed USB device number 10 using xhci_hcd
[17558.770986] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
```

dkraft, 2018-07-30:
I hope this helps. I'm having same issue.  uname -a
Linux 8pack 4.15.0-29-generic #31-Ubuntu SMP Tue Jul 17 15:39:52 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

[ 6834.926322] usb 8-2: new full-speed USB device number 20 using xhci_hcd
[ 6835.335818] usb 8-2: Manufacturer: ODrive Robotics
[ 6835.335819] usb 8-2: SerialNumber: 366533793037
[ 6835.339575] cdc_acm 8-2:1.0: ttyACM0: USB ACM device
[ 6856.742403] usb 8-2: reset full-speed USB device number 20 using xhci_hcd
[ 6857.028139] cdc_acm 8-2:1.0: ttyACM0: USB ACM device
dkraft@8pack:~$ odrivetool -p usb:8:20
ODrive control utility v0.4.1
/usr/local/lib/python3.6/dist-packages/IPython/paths.py:61: UserWarning: IPython dir '/home/dkraft/.ipython' is not a writable location, using a temp directory.
  " using a temp directory.".format(ipdir))
Please connect your ODrive.
You can also type help() or quit().

In [1]: quit

dkraft@8pack:~$ odrivetool -p usb:008:020
ODrive control utility v0.4.1
/usr/local/lib/python3.6/dist-packages/IPython/paths.py:61: UserWarning: IPython dir '/home/dkraft/.ipython' is not a writable location, using a temp directory.
  " using a temp directory.".format(ipdir))
Please connect your ODrive.
You can also type help() or quit().

In [1]: quit

corrupted size vs. prev_size
^[[AAborted (core dumped)
134 dkraft@8pack:~$ odrivetool -p serial:/dev/ttyACM0
ODrive control utility v0.4.1
/usr/local/lib/python3.6/dist-packages/IPython/paths.py:61: UserWarning: IPython dir '/home/dkraft/.ipython' is not a writable location, using a temp directory.
  " using a temp directory.".format(ipdir))
Please connect your ODrive.
You can also type help() or quit().

In [1]: quit

dkraft@8pack:~$ odrivetool -p serial:/dev/ttyACM0
ODrive control utility v0.4.1
/usr/local/lib/python3.6/dist-packages/IPython/paths.py:61: UserWarning: IPython dir '/home/dkraft/.ipython' is not a writable location, using a temp directory.
  " using a temp directory.".format(ipdir))
Please connect your ODrive.
You can also type help() or quit().

In [1]: quit

dkraft@8pack:~$ odrivetool -s 366533793037
ODrive control utility v0.4.1
/usr/local/lib/python3.6/dist-packages/IPython/paths.py:61: UserWarning: IPython dir '/home/dkraft/.ipython' is not a writable location, using a temp directory.
  " using a temp directory.".format(ipdir))
Please connect your ODrive.
You can also type help() or quit().

In [1]: quit

madcowswe, 2018-09-21:
Should be fixed in firmware version 0.4.4

closed 2018-09-21 by madcowswe

Shreeyak, 2018-09-21:
Awesome! Thank you! I can now start with ROS development for the ODrive!