odriverobotics/ODrive #200: `erase_configuration` and power cycling doesn't fully erase configuration

chachi, 2018-07-24:
Occasionally I'll call `erase_configuration`, power cycle the ODrive, run a full recalibration and then when I set a very low velocity setpoint, the motor will spin at what seems like max speed.

I don't have any more of a concrete pattern than that, but it certainly seems to be `erase_configuration`-related.

FW v0.4.1 and HW v3.5

madcowswe, 2018-07-24:
What settings are non-default through an erase_configuration and power cycle?
What settings do you set after the config erase?
What is the effect if you don't erase config but do a power cycle?
Try to read all the configs to see what is different in the different scenarios: just looking at a spinout or not isn't enough information.
Thanks.

Capo01, 2018-09-08:
I have also had similar problems in the past. However, ever since I have been using `odrv0.reboot()` directly after `odrv0.erase_configuration()` the has been no problems. 

For the time being how about changing `odrv0.save_configuration()` and ``odrv0.erase_configuration()` to `odrv0.save_configuration_reboot()` to `odrv0.erase_configuration_reboot() or similar?

samuelsadok, 2020-05-20:
The issue @Capo01 is describing is tracked in #349 and now fixed on `devel`.
It's well possible that that's what @chachi was observing. Since there's been no new information on this for a while I'm gonna close this now. Feel free to open a new issue if this happens again after updating to the latest devel.

closed 2020-05-20 by samuelsadok