odriverobotics/ODrive #202: Race condition between CDC and Native usb endpoints

metanoic, 2018-07-26:
99% certain this relates to #195 and the following forum thread: https://discourse.odriverobotics.com/t/odrive-doesnt-show-up-in-odrivetool/808/27

Brief description of the issue is that while communicating with the board via the "Native" interface, the operating system - while looking for modems or other devices - communicates with the CDC interface and will essentially cause a stall on the native interface.

Ultimately the [`active_endpoint_pair`](https://github.com/madcowswe/ODrive/blob/master/Firmware/communication/interface_usb.cpp#L112) variable is set with the last usb endpoint to have received data.  Then that endpoint is written to and primed for receiving the next packet via [`USBD_CDC_ReceivePacket`](https://github.com/madcowswe/ODrive/blob/master/Firmware/communication/interface_usb.cpp#L102).  If the native endpoint receives a packet and _immediately_ afterwards the CDC receives a packet _before_ we have a chance to respond to the native endpoint, the `active_endpoint_pair` value now corresponds to only the CDC endpoint and the native endpoint never gets primed for its next packet.  The result is a "stalled" connection, or however you want to phrase it.

madcowswe, 2018-09-08:
Sorry I should have made a PR for this, but if anyone is interested in the changeset, it's here:
https://github.com/madcowswe/ODrive/compare/fw-v0.4.3...fb5cae84

camrbuss, 2018-09-11:
Was having this issue on Ubuntu 16.04 and just falshed the devel branch and its working perfect. Thanks Oscar!

madcowswe, 2018-09-21:
Should be fixed in firmware version 0.4.4

closed 2018-09-21 by madcowswe