odriverobotics/ODrive #212: odrivetool rate-test:  'RemoteObject' object has no attribute 'motor0'

Capo01, 2018-08-15:
Running: odrivetool rate-test gives:

```
(C:\Users\R_Desktop\Anaconda3) C:\Users\R_Desktop>odrivetool rate-test
Waiting for ODrive...
reading 10000 values...
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
C:\Users\R_Desktop\Anaconda3\Scripts\odrivetool in <module>()
    188                                               search_cancellation_token=app_shutdown_token,
    189                                               channel_termination_token=app_shutdown_token)
--> 190         rate_test(my_odrive)
    191
    192     elif args.command == 'udev-setup':

C:\Users\R_Desktop\Anaconda3\lib\site-packages\odrive\utils.py in rate_test(device)
    117     vals = []
    118     for _ in range(numFrames):
--> 119         vals.append(device.motor0.loop_counter)
    120
    121     plt.plot(vals)

C:\Users\R_Desktop\Anaconda3\lib\site-packages\fibre\remote_object.py in __getattribute__(self, name)
    205             return attr
    206         else:
--> 207             return object.__getattribute__(self, name)
    208             #raise AttributeError("Attribute {} not found".format(name))
    209

AttributeError: 'RemoteObject' object has no attribute 'motor0'
```

-----------------------------------
The relevant section from utils.py

```
def rate_test(device):
    """
    Tests how many integers per second can be transmitted
    """

    import matplotlib.pyplot as plt
    plt.ion()

    print("reading 10000 values...")
    numFrames = 10000
    vals = []
    for _ in range(numFrames):
        vals.append(device.motor0.loop_counter)

    plt.plot(vals)

    loopsPerFrame = (vals[-1] - vals[0])/numFrames
    loopsPerSec = (168000000/(2*10192))
    FramePerSec = loopsPerSec/loopsPerFrame
    print("Frames per second: " + str(FramePerSec))
```
If someone can tell me what to replace `vals.append(device.motor0.loop_counter)` with I'm happy to make the change.

madcowswe, 2018-08-17:
Thanks, it's `.axis0.loop_counter` now.

Capo01, 2018-08-20:
Change has be made to devel.

closed 2018-08-20 by Capo01