odriverobotics/ODrive #219: Unable to use closed loop control after index only search on startup

Capo01, 2018-08-20:
I am unable to have the motor enter closed loop control after running only an index search. V3.2 board with AMT102 encoder running devel. All other control appears normal.

All settings erased before I entered the following commands into odrivetool as per the instructions:

```
odrv0.axis0.requested_state = AXIS_STATE_FULL_CALIBRATION_SEQUENCE
odrv0.axis0.encoder.config.use_index = True
odrv0.axis0.requested_state = AXIS_STATE_ENCODER_INDEX_SEARCH
odrv0.axis0.requested_state = AXIS_STATE_ENCODER_OFFSET_CALIBRATION
odrv0.axis0.motor.config.pre_calibrated = True
odrv0.axis0.config.startup_encoder_index_search = True
odrv0.axis0.config.startup_closed_loop_control = True
odrv0.save_configuration()
odrv0.reboot()
```

On reboot the motor does its index search but does not enter closed loop control. No error codes and control is in the idle state.
```
In [202]: hex(odrv0.axis0.error)
Out[202]: '0x1'
In [203]: hex(odrv0.axis0.encoder.error)
Out[203]: '0x0'
In [204]: hex(odrv0.axis0.motor.error)
Out[204]: '0x0'
In [205]: odrv0.axis0.motor.is_calibrated
Out[205]: True
```

Adding the following commands resolves the issue
```
odrv0.axis0.config.startup_encoder_offset_calibration = True
odrv0.save_configuration()
odrv0.reboot()
```

Now the motor first does its index search and then does its offset calibration. Only now is closed loop control entered. 

This does not seem to be correct to me as the motor should be usable without the need for the offset calibration once the index signal has been setup. If it is simply an error in the order that I am entering commands then perhaps the instructions may need to be changed.

Wetmelon, 2018-08-22:
I would try:

```
odrv0.axis0.encoder.config.use_index = True
odrv0.axis0.requested_state = AXIS_STATE_FULL_CALIBRATION_SEQUENCE
odrv0.axis0.motor.config.pre_calibrated = True
odrv0.axis0.config.startup_encoder_index_search = True
odrv0.axis0.config.startup_closed_loop_control = True
odrv0.save_configuration()
odrv0.reboot()
```

Capo01, 2018-08-22:
I erased the board and entered the commands in the order you suggested. The problem still persists. On reboot it moves to the index location but then fails to enter closed loop control.

Wetmelon, 2018-08-22:
So you can just move the motor by hand?

Capo01, 2018-08-23:
Thats right, the motor behaves as if the power supply is turned off.

Capo01, 2018-08-23:
I just tried the same set of commands on my V3.5 board and there was no change, the motor shuts off after the index pulse search is completed. This is while running devel btw.

Capo01, 2018-08-23:
Metanoic also confirmed that he sees the same issue on his board.

Wetmelon, 2018-08-24:
`odrv0.axis0.encoder.config.pre_calibrated = True` must also be set after calibration.

The following works for me:
```
odrv0.axis0.encoder.config.use_index = True
odrv0.axis0.requested_state = AXIS_STATE_FULL_CALIBRATION_SEQUENCE
odrv0.axis0.encoder.config.pre_calibrated = True
odrv0.axis0.motor.config.pre_calibrated = True
odrv0.axis0.config.startup_encoder_index_search = True
odrv0.axis0.config.startup_closed_loop_control = True
odrv0.save_configuration()
odrv0.reboot()
```

Capo01, 2018-08-24:
I tried that exact set of comamnds and on reboot it spun up to full speed and then stayed there. Looks like we were missing `odrv0.axis0.requested_state = AXIS_STATE_ENCODER_INDEX_SEARCH` this time. I added that into the mix and now it works perfect. 

```
odrv0.erase_configuration()
odrv0.reboot()
odrv0.axis0.encoder.config.use_index = True
odrv0.axis0.requested_state = AXIS_STATE_FULL_CALIBRATION_SEQUENCE
odrv0.axis0.requested_state = AXIS_STATE_ENCODER_INDEX_SEARCH
odrv0.axis0.encoder.config.pre_calibrated = True
odrv0.axis0.motor.config.pre_calibrated = True
odrv0.axis0.config.startup_encoder_index_search = True
odrv0.axis0.config.startup_closed_loop_control = True
odrv0.save_configuration()
odrv0.reboot()
```

I will add some info to the docs now but otherwise I think thats it.

Capo01, 2018-08-25:
After trying a few things I am still having the same issue. If I run the commands as @Wetmelon suggested my motor will spin up uncontrollably after restart. It does appear to do the index search after calling `AXIS_STATE_FULL_CALIBRATION_SEQUENCE` but this is not applied when using `odrv0.axis0.encoder.config.pre_calibrated = True`. 

```In [20]: odrv0.axis0.encoder.config.use_index = True

In [21]: odrv0.axis0.requested_state = AXIS_STATE_FULL_CALIBRATION_SEQUENCE

In [22]: odrv0.axis0.encoder.index_found
Out[22]: True

In [23]: odrv0.axis0.encoder.phase
Out[23]: -0.009570121765136719

In [24]: odrv0.axis0.encoder.pos_estimate
Out[24]: -493.08349609375```

This looks to be an issue with the order in which the startup calibration is conducted. If I replace the following in axis.cpp:

```            } else if (requested_state_ == AXIS_STATE_FULL_CALIBRATION_SEQUENCE) {
                task_chain_[pos++] = AXIS_STATE_MOTOR_CALIBRATION;
                if (encoder_.config_.use_index)
                    task_chain_[pos++] = AXIS_STATE_ENCODER_INDEX_SEARCH;
                task_chain_[pos++] = AXIS_STATE_ENCODER_OFFSET_CALIBRATION;
                task_chain_[pos++] = AXIS_STATE_IDLE;```

with:

```            } else if (requested_state_ == AXIS_STATE_FULL_CALIBRATION_SEQUENCE) {
                task_chain_[pos++] = AXIS_STATE_MOTOR_CALIBRATION;
                task_chain_[pos++] = AXIS_STATE_ENCODER_OFFSET_CALIBRATION;
                if (encoder_.config_.use_index)
                    task_chain_[pos++] = AXIS_STATE_ENCODER_INDEX_SEARCH;
                task_chain_[pos++] = AXIS_STATE_IDLE;```

The index position is now applied when ` AXIS_STATE_FULL_CALIBRATION_SEQUENCE` is called. Pull request [here](https://github.com/madcowswe/ODrive/pull/225).

@Wetmelon are you able to confirm that works for you?

madcowswe, 2018-08-25:
@Wetmelon's series of commands listed in [this comment](https://github.com/madcowswe/ODrive/issues/219#issuecomment-415627985) is expected to work, and would be the exact sequence of commands I'd have used myself. I'll give it a go a few times here.

For reference, what commit are you running @Capo01?

madcowswe, 2018-08-25:
It seems to work just fine for me on 346a98b. Running D5065 motors and AMT102 encoder. Command transcript:
```py
Connected to ODrive 206E36903548 as odrv0
In [1]: odrv0.erase_configuration()

In [2]: odrv0.reboot()
---------------------------------------------------------------------------
ChannelBrokenException                    Traceback (most recent call last)
C:\Users\Oskar\Documents\ODrive\Firmware\fibre\python\fibre\shell.py in <module>()
----> 1 odrv0.reboot()

C:\Users\Oskar\Documents\ODrive\Firmware\fibre\python\fibre\remote_object.py in __call__(self, *args)
    160         for i in range(len(args)):
    161             self._inputs[i].set_value(args[i])
--> 162         self._parent.__channel__.remote_endpoint_operation(self._trigger_id, None, True, 0)
    163         if len(self._outputs) > 0:
    164             return self._outputs[0].get_value()

C:\Users\Oskar\Documents\ODrive\Firmware\fibre\python\fibre\protocol.py in remote_endpoint_operation(self, endpoint
_id, input, expect_ack, output_length)
    312                     return self._responses.pop(seq_no)
    313                     # TODO: record channel statistics
--> 314                 raise ChannelBrokenException() # Too many resend attempts
    315             finally:
    316                 self._expected_acks.pop(seq_no)

ChannelBrokenException:

Reconnected to ODrive 206E36903548 as odrv0
In [3]: odrv0.axis0.encoder.config.use_index = True
   ...: odrv0.axis0.requested_state = AXIS_STATE_FULL_CALIBRATION_SEQUENCE
   ...:

In [4]:

In [4]: odrv0.axis0.encoder.config.pre_calibrated = True
   ...: odrv0.axis0.motor.config.pre_calibrated = True
   ...: odrv0.axis0.config.startup_encoder_index_search = True
   ...: odrv0.axis0.config.startup_closed_loop_control = True
   ...:

In [5]: odrv0.save_configuration()
   ...: odrv0.reboot()
   ...:
---------------------------------------------------------------------------
ChannelBrokenException                    Traceback (most recent call last)
C:\Users\Oskar\Documents\ODrive\Firmware\fibre\python\fibre\shell.py in <module>()
      1 odrv0.save_configuration()
----> 2 odrv0.reboot()

C:\Users\Oskar\Documents\ODrive\Firmware\fibre\python\fibre\remote_object.py in __call__(self, *args)
    160         for i in range(len(args)):
    161             self._inputs[i].set_value(args[i])
--> 162         self._parent.__channel__.remote_endpoint_operation(self._trigger_id, None, True, 0)
    163         if len(self._outputs) > 0:
    164             return self._outputs[0].get_value()

C:\Users\Oskar\Documents\ODrive\Firmware\fibre\python\fibre\protocol.py in remote_endpoint_operation(self, endpoint
_id, input, expect_ack, output_length)
    312                     return self._responses.pop(seq_no)
    313                     # TODO: record channel statistics
--> 314                 raise ChannelBrokenException() # Too many resend attempts
    315             finally:
    316                 self._expected_acks.pop(seq_no)

ChannelBrokenException:

Reconnected to ODrive 206E36903548 as odrv0
In [6]: odrv0.axis0.controller.pos_setpoint = 100000

In [7]: #motor spins fine here

In [8]:
```

madcowswe, 2018-08-25:
Actually, when I now
1. turn off power
2. move rotor to random position by hand
3. apply power

It searches for the idx, then immediatly spins out!
OK I'll investigate.

madcowswe, 2018-08-25:
I found the bug I think. Check the newest devel (commit 584b739).

Wetmelon, 2018-08-25:
Can confirm I'm experiencing the same bug.  Investigating the change that @madcowswe implemented.

Capo01, 2018-08-31:
I have tested the change and its now working as intended. I think that solves it then.

Capo01, 2018-08-31:
Further playing around and I see a new issue. If the motor is disabled due to an error (e.g. `ERROR_DC_BUS_OVER_VOLTAGE = 0x04`) and I use `odrv0.reboot()` or power cycle then the motor will begin to spin up to full speed. If I then restart again control returns to normal.

madcowswe, 2018-09-01:
Do you mind to describe this new issue in a new "issue"? Then I can close this one, and we can focus on that in a new one.

closed 2018-09-01 by madcowswe