odriverobotics/ODrive #221: Anti-cogging calibration process becomes stuck

Capo01, 2018-08-21:
During the anti-cogging calibration process the motor becomes stuck and oscillates back and forth.

Running V3.2 board on a 24V, 14.6A PSU with devel with a AMT102 encoder on a SK3 280KV motor on axis 0 with nothing attached to the motor.

Gains and current limits are as follows:
```
odrv0.axis0.motor.config.current_lim = 40
odrv0.axis0.controller.config.vel_limit = 200000
odrv0.axis0.controller.config.pos_gain = 40.0
odrv0.axis0.controller.config.vel_gain = 3.0 / 10000.0
odrv0.axis0.controller.config.vel_integrator_gain = 20.0 / 10000.0
```

Index search is undertaken during startup.

I started liveplotter during the anti-cogging calibration process and see the following:

[image]

It appears that the calibration process is working fine until around 600 counts where it becomes stuck. I have left this run for around 10 min and it does not get past this position.

No errors are reported while its in this stuck state.

Wetmelon, 2018-08-22:
Looks like a gains issue.  You want to bump your gains up quite a bit when doing the calibration.  I was running the following gains for really good 

pos_gain = 200.0
vel_gain = 0.0005
vel_integrator_gain = 0.0150

This was for the stock N5065 motors with AMT203 encoder at 4096 CPR

Capo01, 2018-08-22:
I tried increasing my gains to the values you recommended but doing so made the system unstable (vibration of +- 10 counts) and nothing happens when after entering `start_anticogging_calibration()`. Dropping the gains back down to the value I was using before and anti-cogging calibration now starts again.

I tried the opposite extreme and dropped all  the gains down to their default values and again ran anti-cogging. The calibration takes longer this time but the rotor again become stuck, this time at a different position.

[image]

 I also tried reducing the current limit to its default value but this also had no effect.

Capo01, 2018-08-22:
I should also mention that the point at which it becomes stuck is very repeatable for a fixed set of gains.

Capo01, 2018-08-23:
Also tried this on my V3.5 board with the D6374 motors, 48V PSU firmware. No change in the behaviour, it still gets stuck.

Capo01, 2018-08-23:
Metanoic also gave this a go (AMT102 encoder) and says he is getting the same result. Is it worth testing with a different encoder?

Wetmelon, 2018-08-23:
It might be worth a shot.  Adjusting gains manually can also help.  It's more or less expected that the motor will oscillate over a point with incorrect gains, and the required gains change as the motor rotates due to the cogging torque changing the system.

You can also try adjusting `odrv0.axis0.controller.anticogging.calib_pos_threshold` and `odrv0.axis0.controller.anticogging.calib_vel_threshold`

These values are 1 by default, this means anticogging will only proceed with < +/- 1 count error, and < +/- 1 count per sec velocity error.  Bumping them up slightly can help.

Wetmelon, 2018-09-02:
@metanoic was able to get his anticogging to work.  Did yours ever behave?

Akex2, 2018-09-03:
other issue ? 
https://youtu.be/NZE-tJz-bXw

madcowswe, 2020-08-30:
We plan to replace the current anti-cogging implementation with one that keeps rolling rather than try to stop at every sample.

closed 2020-08-30 by madcowswe