odriverobotics/ODrive #226: Incremental / circular position control mode.

madcowswe, 2018-08-25:
Allow the motor to be incrementally advanced forever in a incremental / circular control mode.
This should allow continuous spinning in step/dir input mode.

madcowswe, 2018-10-05:
Note that instead of a fixed circular space, we could do a "reference shift on demand", as I described [here](https://discourse.odriverobotics.com/t/incremental-movement-in-position-mode/1167/5?u=madcowswe)

Wetmelon, 2019-09-08:
I believe this is ALMOST working, except shadow count will eventually roll over?

samuelsadok, 2020-05-20:
This is actually implemented since firmware v0.4.7.
A rollover of shadow_count shouldn't pose any issue since it's only used to calculate the delta between two encoder updates.

closed 2020-05-20 by samuelsadok