odriverobotics/ODrive #230: PWM Motor Control

abickel, 2018-09-01:
Please create a PWM motor control function (0 to 100% duty cycle).
My specific application is spindle motor for a CNC Mill and the motion controller (Ethernet Smooth Stepper) outputs a 0~100% duty cycle at a user specified frequency.  In addition, the ODrive needs to accept a motor direction input (single pin).

Many thanks.

Wetmelon, 2018-09-02:
#65 addressed this to some extent.  Looks like it's more like an RC-style PWM input.  @madcowswe will know more.

madcowswe, 2018-10-11:
Also known common mode:
* "0 to 100" -> "0 to max"
* "0 50 100" -> "-max 0 +max"

robsonswiss, 2018-10-12:
Just to clarify, "0 to 100" goes with a direction input as well.

Wetmelon, 2021-09-13:
Won't add, but we accept well-tested PRs.

closed 2021-09-13 by Wetmelon