odriverobotics/ODrive #231: Read UART on interrupt

Nate711, 2018-09-01:
Currently there's a worst-case delay of 1ms between receiving a uart message and actually reading and acting upon it. To fix this, the odrive could trigger an interrupt when it detects a new uart byte, or better, when it detects that a string of bytes have finished sending (idle line interrupt). 

The idle line interrupt can be set up like this:
```cpp
void start_uart_server() {
    __HAL_UART_ENABLE_IT(&huart4, UART_IT_IDLE);   // enable idle line interrupt
   // rest of code...
}
```

And then you can handle the interrupt like so:
```cpp
void UART4_IRQHandler(void)
  /* USER CODE BEGIN UART4_IRQn 0 */
  if(huart4.Instance->SR & UART_FLAG_IDLE) {
    // Trigger the uart server thread to read the dma 
  }

  /* USER CODE END UART4_IRQn 0 */
  // HAL_UART_IRQHandler(&huart4);
  /* USER CODE BEGIN UART4_IRQn 1 */

  /* USER CODE END UART4_IRQn 1 */
}
```

What I wasn't able to get to work was clearing the interrupt flags successfully. Inside the if statement above I had the code:
```cpp
volatile uint32_t tmp;                                   /* Must be volatile to prevent optimizations */
tmp = huart4.Instance->SR;                       /* Read status register */
tmp = huart4.Instance->DR;
if(uart_thread) {
   osSignalSet(uart_thread, 0x01);
}
```
But if I remember correctly the flags were not being cleared successfully.

I also set up the uart server thread to wait on a thread signal like so:
```cpp
evt = osSignalWait(0x01, osWaitForever); // block until interrupt sets the flag
if (evt.status == osEventSignal) { // only query dma if got an event signal 
}
```

sgilbert182, 2019-12-09:
Hi, I found with the UART on the STM32's it was very effective to have a 2 byte wide circular receive buffer tied using DMA to the UART. Then have the DMA interrupt on every half and full receive. In that interrupt you can place the received data in your character queue. This gives you a nice double buffering of the UART and no need to wait for the idle interrupt plus as the DMA clears the UART receive buffer, you have a lot longer to process the incoming data.

sgilbert182, 2019-12-09:
It's also worth noting, the UART processing task has a periodic wake up of 1ms and is suspended in the meantime. That could be changed to a sleep/suspend and wake on UART interrupt giving much faster response.

madcowswe, 2020-08-30:
Fixed on 6fa76e1 by increasing UART polling frequency to 8kHz.

closed 2020-08-30 by madcowswe