odriverobotics/ODrive #232: Uncontrolled motor spinup when using only index pulse calibration

Capo01, 2018-09-01:
If the motor is disabled due to an error (e.g. ERROR_DC_BUS_OVER_VOLTAGE = 0x04) and I use odrv0.reboot() or power cycle then the motor will begin to spin up to full speed. If I then restart again control returns to normal.

If I power-cycle or reboot without an error condition then the startup is normal. I have also noticed that if I power cycle over and over at short intervals I can cause the spinup problem.

Capo01, 2018-09-08:
Ok, sorry for the delay. V3.2 board running of 14.8A PSU, 280KV SK3 motor with 8192 AMT102 encoder. Flywheel attached to the motor. Running current devel I enter the following:

```odrv0.erase_configuration()
odrv0.reboot()

odrv0.config.brake_resistance = 0.53
odrv0.axis0.encoder.config.use_index = True
odrv0.axis0.requested_state = AXIS_STATE_FULL_CALIBRATION_SEQUENCE

odrv0.axis0.encoder.config.pre_calibrated = True
odrv0.axis0.motor.config.pre_calibrated = True
odrv0.axis0.config.startup_encoder_index_search = True
odrv0.axis0.config.startup_closed_loop_control = True
odrv0.axis0.motor.config.current_lim = 40
odrv0.save_configuration()

odrv0.reboot()```

After multiple reboots and power cycles there was no uncontrolled spin-up. 

However, if I do the following then the motor will spin-up uncontrollably on startup:

1. Remove power
2. Spin the motor by hand in the opposite direction to that used for the index pulse search
3. Restore power

This also works if I reboot and quickly spin it by hand. No such problem if I spin it by hand in the same direction as that used for the index pulse search. Note that this is with a large flywheel attached to the motor.

madcowswe, 2018-09-21:
Can you check if the issue persists in firmware version 0.4.4?

Wetmelon, 2019-03-25:
@Capo01 Reminder to verify issue.

Capo01, 2019-03-25:
Thanks for the reminder. I will take a look this weekend.

robsonswiss, 2019-05-17:
@Capo01 did you check for a false index? I'm seeing similar condition with false index found even though the index line is clean on the scope.

madcowswe, 2020-08-29:
Continued in #462.

closed 2020-08-29 by madcowswe