odriverobotics/ODrive #240: Use a hall signal as index.

madcowswe, 2018-09-10:
Use a "preroll" manouver, that will start rotating the motor, and only after it is rotating in a known direction activate the sensitivity to the index rising edge. This should let you use a hall input as a makeshift index pulse.

neomanic, 2019-02-28:
Did this break in one of the merges, or perhaps an extra variable is now required? (https://github.com/madcowswe/ODrive/commit/e3785a2aa24a9079da460858fd02498747671987 maybe?)

Reason being; we have been running the original preroll you released to us in d7644daacaf5e58114bd7832da54850fcaf69e68, but then after updating to 1178bf4412cddc71b592c12f5db20be0e2c33f2e and running the commission script with the variable name changes (find_idx_on_lockin -> find_idx_on_lockin_only, ax.config.lockin_* -> ax.config.lockin.*), now when first powering up the motor and running the AXIS_STATE_ENCODER_INDEX_SEARCH, it will run indefinitely in the index search.

madcowswe, 2019-03-05:
@neomanic Is this still an issue (I know we did some back and forth to figure out some other issues)?

neomanic, 2019-03-05:
Yes, still an issue. We decided to leave the ODrive-related issues alone last week since we had other areas to address. I'm back home this week, so I'll update firmware to the latest devel tomorrow and give it a whirl.

closed 2020-08-29 by madcowswe