odriverobotics/ODrive #256: Linear Motor Support

jamescarter-le, 2018-10-03:
Hi,

  Your drive looks very nice - but I'm wondering if the control software can drive a brushless DC linear motor.

  I have a linear shaft motor with no hall sensor, so needs an initial commutation alignment, but with an incremental encoder is able to commutate effectively.

 Does ODrive support this type of motor?

Thanks,
James

madcowswe, 2018-10-06:
Yes, I think that should work just like the same situation with a rotary motor.

Wetmelon, 2018-10-07:
Please let us know how your build goes.  It's been talked about quite a bit, and we believe it will work, but I don't think anyone has actually completed a linear motor test.  https://discourse.odriverobotics.com/

jamescarter-le, 2018-10-07:
Thanks - I have 3 Tecnotion linear motors which are currently being run by Kollmorgen AKD servo drives.

I am considering 4 linear shaft motors (http://www.nipponpulse.com/products/overview/linear-shaft-servomotors) for a set of axis required in the machine and need a cost competitive way of driving them.

Is there anywhere I can buy a sample ODrive populated board for me to investigate if this would be a suitable option?

I am intending on pairing these with RLS linear encoders, so it would be useful to know the max frequency the ODrive can read the A/B pulses as this determines which encoder resolution to source.

madcowswe, 2018-10-07:
Sure, you can buy ODrives [here](https://odriverobotics.com/shop).

We haven't tested to the encoder speed limit (we don't have a fast enough encoder). However the input filtering settings on the ODrive by default should allow signals up to 3.5MHz, so about 14 Mcounts/s. If you need more speed than that, we can reconfigure the filters, which allows a theoretical max frequency of 21MHz (84 Mcount/s).

Wetmelon, 2019-03-25:
@jamescarter-le did you end up testing this?

jamescarter-le, 2019-03-25:
No I ended up using specific servo drivers from Kollmorgen. For this
application the cost was not the main driver and so these servo drives are
able to drive linear motors natively, and also communicate with the linear
encoders. Also the built in feedback across drives helps drives gantries
very easily.

Wetmelon, 2019-03-25:
Ok, I'm going to close this issue for now.  If you end up trying the drivers out let us know how it goes.

closed 2019-03-25 by Wetmelon