odriverobotics/ODrive #272: torque/current mode problem

jiangj001, 2018-11-16:
Hi, when I'm using current mode, no matter what current value I set, the motor just keeps spinning faster and faster, until hitting the speed limit. 
During the process, my power supply also showed the current started from 0.1A, and went all the way to 0.36 A.
I think the current control mode is supposed to consume the same current from the power supply, is that correct? If not, did I do something wrong?

Wetmelon, 2018-11-17:
> Hi, when I'm using current mode, no matter what current value I set, the motor just keeps spinning faster and faster, until hitting the speed limit.

Yeah, that's typical.   You don't have a load on the motor so as soon as it overcomes stiction, it will accelerate to max speed.

> I think the current control mode is supposed to consume the same current from the power supply, is that correct?

That is not correct.  The current drawn from the power supply will change depending on the speed of the motor.  In short, the *power* drawn by the motor and supplied by the PSU will match.  The PSU is constant-voltage, the motor is constant-current.

elektrinis, 2018-12-06:
This is completely normal. Constant current = constant torque. If torque is higher than friction and load, it will accelerate. So if there is no load, it will accelerate to max speed.
If all is working as designed, current is limited on motor (which provides constant torque). Since power = speed * torque, and speed is increasing, you will see increased power consumption from power supply when at higher speed.

closed 2019-01-04 by madcowswe