odriverobotics/ODrive #283: Unable to add integer config value

Wetmelon, 2018-12-24:
When I add an integer or integer-like (e.g. an enum) to the controller config, protocol refuses to compile.

controller.hpp config struct:
```
    struct Config_t {
        ControlMode_t control_mode = CTRL_MODE_POSITION_CONTROL;  //see: Motor_control_mode_t
        float pos_gain = 20.0f;  // [(counts/s) / counts]
        float vel_gain = 5.0f / 10000.0f;  // [A/(counts/s)]
        // float vel_gain = 5.0f / 200.0f, // [A/(rad/s)] <sensorless example>
        float vel_integrator_gain = 10.0f / 10000.0f;  // [A/(counts/s * s)]
        float vel_limit = 20000.0f;        // [counts/s]
        float vel_limit_tolerance = 1.2f;  // ratio to vel_lim. 0.0f to disable
        float vel_ramp_rate = 10000.0f;  // [(counts/s) / s]
        bool setpoints_in_cpr = false;
        int something = 3;
    };
```
Controller.hpp make_protocol_cobject for config:
```
            make_protocol_object("config",
                make_protocol_property("control_mode", &config_.control_mode),
                make_protocol_property("pos_gain", &config_.pos_gain),
                make_protocol_property("vel_gain", &config_.vel_gain),
                make_protocol_property("vel_integrator_gain", &config_.vel_integrator_gain),
                make_protocol_property("vel_limit", &config_.vel_limit),
                make_protocol_property("vel_limit_tolerance", &config_.vel_limit_tolerance),
                make_protocol_property("vel_ramp_rate", &config_.vel_ramp_rate),
                make_protocol_property("setpoints_in_cpr", &config_.setpoints_in_cpr),
                make_protocol_property("something", &config_.something)
            ),
```
Error:
```
fibre/cpp/include/fibre/protocol.hpp:892:25: error: call of overloaded 'to_string(int&, char*&, size_t&, int)' is ambiguous
         return to_string(*property_, buffer, length, 0);
                ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
fibre/cpp/include/fibre/protocol.hpp:646:13: note: candidate: bool to_string(const float&, char*, size_t, int) [with T = float; size_t = unsigned int]
 static bool to_string(const float& value, char * buffer, size_t length, int) {
             ^~~~~~~~~
fibre/cpp/include/fibre/protocol.hpp:651:13: note: candidate: bool to_string(const bool&, char*, size_t, int) [with T = bool; size_t = unsigned int]
 static bool to_string(const bool& value, char * buffer, size_t length, int) {
             ^~~~~~~~~
fibre/cpp/include/fibre/protocol.hpp:657:13: note: candidate: bool to_string(const T&, char*, size_t, ...) [with T = int; size_t = unsigned int]
 static bool to_string(const T& value, char * buffer, size_t length, ...) {
             ^~~~~~~~~
```

WallyWalrick, 2019-01-01:
I had this issue but found int32_t works
Think it's because int is too ambiguous

madcowswe, 2019-01-04:
@Wetmelon can you try int32_t?

Wetmelon, 2019-03-25:
Yeah, specific types work.  Apparently there's a conflict somewhere due to an implicit conversion.  Will just avoid it

closed 2019-03-25 by Wetmelon