odriverobotics/ODrive #288: libc++abi.dylib: terminating with uncaught exception of type NSException

Luffy4321, 2019-01-08:
command  odrivetool liveplotter  is wrong ,as follows:
`
 ✘  ~  odrivetool liveplotter
Waiting for ODrive...
Showing plot. Press Ctrl+C to exit.
2019-01-08 17:23:30.342 Python[1309:12151] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow drag regions should only be invalidated on the Main Thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff32881e65 __exceptionPreprocess + 256
	1   libobjc.A.dylib                     0x00007fff5e8d8720 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff3289b8e5 -[NSException raise] + 9
	3   AppKit                              0x00007fff2fda07e6 -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 324
	4   AppKit                              0x00007fff2fd9dbd0 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1488
	5   AppKit                              0x00007fff2fd9d5fa -[NSWindow initWithContentRect:styleMask:backing:defer:] + 45
	6   _macosx.cpython-37m-darwin.so       0x0000000114caa4c0 -[Window initWithContentRect:styleMask:backing:defer:withManager:] + 80
	7   _macosx.cpython-37m-darwin.so       0x0000000114cada17 FigureManager_init + 327
	8   Python                              0x0000000106f600ee wrap_init + 12
	9   Python                              0x0000000106f2779f wrapperdescr_call + 337
	10  Python                              0x0000000106f21a39 _PyObject_FastCallKeywords + 359
	11  Python                              0x0000000106fb765e call_function + 746
	12  Python                              0x0000000106fb0231 _PyEval_EvalFrameDefault + 7016
	13  Python                              0x0000000106f21fae function_code_fastcall + 112
	14  Python                              0x0000000106f22931 _PyObject_Call_Prepend + 150
	15  Python                              0x0000000106f6005c slot_tp_init + 80
	16  Python                              0x0000000106f5cd28 type_call + 178
	17  Python                              0x0000000106f21a39 _PyObject_FastCallKeywords + 359
	18  Python                              0x0000000106fb765e call_function + 746
	19  Python                              0x0000000106fb0231 _PyEval_EvalFrameDefault + 7016
	20  Python                              0x0000000106f21fae function_code_fastcall + 112
	21  Python                              0x0000000106fb7665 call_function + 753
	22  Python                              0x0000000106fb0231 _PyEval_EvalFrameDefault + 7016
	23  Python                              0x0000000106fb7ef7 _PyEval_EvalCodeWithName + 1835
	24  Python                              0x0000000106f21801 _PyFunction_FastCallDict + 441
	25  Python                              0x0000000106f22931 _PyObject_Call_Prepend + 150
	26  Python                              0x0000000106f21ce1 PyObject_Call + 136
	27  Python                              0x0000000106fb0548 _PyEval_EvalFrameDefault + 7807
	28  Python                              0x0000000106fb7ef7 _PyEval_EvalCodeWithName + 1835
	29  Python                              0x0000000106f21b98 _PyFunction_FastCallKeywords + 225
	30  Python                              0x0000000106fb7665 call_function + 753
	31  Python                              0x0000000106fb0231 _PyEval_EvalFrameDefault + 7016
	32  Python                              0x0000000106fb7ef7 _PyEval_EvalCodeWithName + 1835
	33  Python                              0x0000000106f21801 _PyFunction_FastCallDict + 441
	34  Python                              0x0000000106fb0548 _PyEval_EvalFrameDefault + 7807
	35  Python                              0x0000000106f21fae function_code_fastcall + 112
	36  Python                              0x0000000106fb7665 call_function + 753
	37  Python                              0x0000000106fb0218 _PyEval_EvalFrameDefault + 6991
	38  Python                              0x0000000106f21fae function_code_fastcall + 112
	39  Python                              0x0000000106fb7665 call_function + 753
	40  Python                              0x0000000106fb0218 _PyEval_EvalFrameDefault + 6991
	41  Python                              0x0000000106f21fae function_code_fastcall + 112
	42  Python                              0x0000000106f22931 _PyObject_Call_Prepend + 150
	43  Python                              0x0000000106f21ce1 PyObject_Call + 136
	44  Python                              0x000000010701eeee t_bootstrap + 71
	45  Python                              0x0000000106fe5847 pythread_wrapper + 25
	46  libsystem_pthread.dylib             0x00007fff5fb99339 _pthread_body + 126
	47  libsystem_pthread.dylib             0x00007fff5fb9c2a7 _pthread_start + 70
	48  libsystem_pthread.dylib             0x00007fff5fb98445 thread_start + 13
)
libc++abi.dylib: terminating with uncaught exception of type NSException
[1]    1309 abort      odrivetool liveplotter
`

Luffy4321, 2019-01-08:
```
odrivetool liveplotter
Waiting for ODrive...
Showing plot. Press Ctrl+C to exit.
2019-01-08 17:23:30.342 Python[1309:12151] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow drag regions should only be invalidated on the Main Thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff32881e65 __exceptionPreprocess + 256
	1   libobjc.A.dylib                     0x00007fff5e8d8720 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff3289b8e5 -[NSException raise] + 9
	3   AppKit                              0x00007fff2fda07e6 -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 324
	4   AppKit                              0x00007fff2fd9dbd0 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1488
	5   AppKit                              0x00007fff2fd9d5fa -[NSWindow initWithContentRect:styleMask:backing:defer:] + 45
	6   _macosx.cpython-37m-darwin.so       0x0000000114caa4c0 -[Window initWithContentRect:styleMask:backing:defer:withManager:] + 80
	7   _macosx.cpython-37m-darwin.so       0x0000000114cada17 FigureManager_init + 327
	8   Python                              0x0000000106f600ee wrap_init + 12
	9   Python                              0x0000000106f2779f wrapperdescr_call + 337
	10  Python                              0x0000000106f21a39 _PyObject_FastCallKeywords + 359
	11  Python                              0x0000000106fb765e call_function + 746
	12  Python                              0x0000000106fb0231 _PyEval_EvalFrameDefault + 7016
	13  Python                              0x0000000106f21fae function_code_fastcall + 112
	14  Python                              0x0000000106f22931 _PyObject_Call_Prepend + 150
	15  Python                              0x0000000106f6005c slot_tp_init + 80
	16  Python                              0x0000000106f5cd28 type_call + 178
	17  Python                              0x0000000106f21a39 _PyObject_FastCallKeywords + 359
	18  Python                              0x0000000106fb765e call_function + 746
	19  Python                              0x0000000106fb0231 _PyEval_EvalFrameDefault + 7016
	20  Python                              0x0000000106f21fae function_code_fastcall + 112
	21  Python                              0x0000000106fb7665 call_function + 753
	22  Python                              0x0000000106fb0231 _PyEval_EvalFrameDefault + 7016
	23  Python                              0x0000000106fb7ef7 _PyEval_EvalCodeWithName + 1835
	24  Python                              0x0000000106f21801 _PyFunction_FastCallDict + 441
	25  Python                              0x0000000106f22931 _PyObject_Call_Prepend + 150
	26  Python                              0x0000000106f21ce1 PyObject_Call + 136
	27  Python                              0x0000000106fb0548 _PyEval_EvalFrameDefault + 7807
	28  Python                              0x0000000106fb7ef7 _PyEval_EvalCodeWithName + 1835
	29  Python                              0x0000000106f21b98 _PyFunction_FastCallKeywords + 225
	30  Python                              0x0000000106fb7665 call_function + 753
	31  Python                              0x0000000106fb0231 _PyEval_EvalFrameDefault + 7016
	32  Python                              0x0000000106fb7ef7 _PyEval_EvalCodeWithName + 1835
	33  Python                              0x0000000106f21801 _PyFunction_FastCallDict + 441
	34  Python                              0x0000000106fb0548 _PyEval_EvalFrameDefault + 7807
	35  Python                              0x0000000106f21fae function_code_fastcall + 112
	36  Python                              0x0000000106fb7665 call_function + 753
	37  Python                              0x0000000106fb0218 _PyEval_EvalFrameDefault + 6991
	38  Python                              0x0000000106f21fae function_code_fastcall + 112
	39  Python                              0x0000000106fb7665 call_function + 753
	40  Python                              0x0000000106fb0218 _PyEval_EvalFrameDefault + 6991
	41  Python                              0x0000000106f21fae function_code_fastcall + 112
	42  Python                              0x0000000106f22931 _PyObject_Call_Prepend + 150
	43  Python                              0x0000000106f21ce1 PyObject_Call + 136
	44  Python                              0x000000010701eeee t_bootstrap + 71
	45  Python                              0x0000000106fe5847 pythread_wrapper + 25
	46  libsystem_pthread.dylib             0x00007fff5fb99339 _pthread_body + 126
	47  libsystem_pthread.dylib             0x00007fff5fb9c2a7 _pthread_start + 70
	48  libsystem_pthread.dylib             0x00007fff5fb98445 thread_start + 13
)
libc++abi.dylib: terminating with uncaught exception of type NSException
[1]    1309 abort      odrivetool liveplotter

```

Luffy4321, 2019-01-08:
how to fix it ?

madcowswe, 2019-01-10:
I think it's a fundamental issue with how the liveplotter is implemented: we draw on a not-main tread which is not actually allowed. It seems to work anyway on Windows and Linux. But looks like not Mac. Sorry about that. We intend to fix it when we have time. You can try fixing it yourself if you like: you need to move the rendering to the main thread. This probably requires the Multiprocessing module to allow you to keep sending commands.

closed 2021-09-13 by Wetmelon