odriverobotics/ODrive #308: Setting udev rules can fail for more reasons

chachi, 2019-02-02:
Currently the setup.py udev rules configuration only allows for a `PermissionError` to pass (with a warning) and any other error (e.g. `CalledProcessError`) will fail. This means that if `udevadm control --reload-rules` fails for some reason, it is not possible to install `odrivetool`.

Relevant line:
https://github.com/madcowswe/ODrive/blob/451e79519637fdcf33f220f7dae9a28b15e014ba/tools/setup.py#L98

Workaround: make the udev rule immutable/unwriteable to force a `PermissionError` instead of another error, but this is pretty awkward.

madcowswe, 2019-04-02:
Thanks for reporting this. I will change it to catch any exception and hence print a warning. Any rules related failure should still allow (degraded) installation.

madcowswe, 2019-04-03:
Should be fixed on commit 035c899.

closed 2019-04-03 by madcowswe