odriverobotics/ODrive #330: CRC16 Implementation Documentation

jcsoo, 2019-04-18:
Comments in the protocol implementation suggest that a standard CRC16 calculation is being used with the polynomial `0x3D65` (typically used in CRC-16-DNP):

https://github.com/madcowswe/ODrive/blob/4e039e160416f52195e2199cec62a92fa2f103b6/Firmware/fibre/python/fibre/protocol.py#L64-L66

However, running the commented code produces `0x9a3a`:

```
>>> hex(fibre.protocol.calc_crc16(0xfeef, [1, 2, 3, 4, 5, 0x10, 0x13, 0x37]))
'0x9a3a'
```
which is a different result than what the referenced [online calculator](
http://www.sunshine2k.de/coding/javascript/crc/crc_js.html) generates when using the the same polynomial and initial value:

[image]

It probably doesn't make sense to change the algorithm at this point, but if it really is different, it should be documented so that driver implementers know that they have to reproduce this specific algorithm rather than using an existing CRC library.

It would also be handy to have a few test cases to make it easier to confirm correct implementation.

metanoic, 2019-06-07:
Existing CRC libraries should be capable of handling this fine.  You need to untick the input and result reflection options in the `CRC detailed parameters` section.  Doing so will yield `0x9A3A` as the result.

samuelsadok, 2020-05-26:
I clarified the CRC documentation a bit now. See: https://github.com/madcowswe/ODrive/blob/devel/docs/protocol.md#crc-algorithms

closed 2020-08-29 by madcowswe