odriverobotics/ODrive #350: Users would like to control bus voltage when ODrive is being used as a generator

Wetmelon, 2019-05-25:


Wetmelon, 2020-04-26:
This is significantly improved in RazorsEdge, which implements the concept of a "nominal" bus voltage

samuelsadok, 2020-08-31:
I suggest we look at this in the context of a more generalized power solver approach:
Each component (DC, M0, M1, AUX) has certain varying soft and hard constraints regarding their source and sink power capabilities. A general power manager would then look at those constraints and their priorities to choose the final power setpoint for each component such that they sum up to 0.
However in reality we'd have to consider stability concerns and non-convex problems. For example reducing regen power requires to briefly _increase_ regen power in order to get energy out of the coils and thereby reduce regen force.