odriverobotics/ODrive #356: find_any() takes a long time to connect

auxym, 2019-06-03:
I'm not sure whether this is an issue with fibre or odrive's firmware.

However, when initiating communication with the ODrive using the python interface, either using `find_any()` or `odrivetool`, there is a 35 second delay on my machine.

In case this is platform specific, here are some details:

*Windows 10
*Python 3.7.3
*odrive python module v0.4.9
*odrive FW 0.4.10

adammunich, 2019-06-04:
we know it is a problem, the JSON schema takes a long time to download. it is on the todo list to make the transport layer faster to fix this properly.

auxym, 2019-06-05:
Good to know it's not a weird bug I'm the only one to experience 😛 I did not find another issue or discourse conversation on this issue.

In my opinion it's not a major issue, but fast startup would definitely be a nice convenience/UX boost!

zwn, 2019-09-30:
@adammunich Where is the todo list for odrive development? I was checking https://github.com/madcowswe/ODrive/projects/1 but there does not seem to be anything related to json schema download.

zwn, 2019-10-04:
I am seeing only about 6s delay in odrivetool from the time I power up the odrive to the time when the tool reports it has found it.

jhurliman, 2020-03-16:
For additional context, this takes a long time because the JSON schema is approximately ~27KB but the controller is only send 32 byte responses (30 bytes of JSON payload), so the schema is downloaded over 900+ round trips. Each round trip is quite slow compared to other commands, taking several milliseconds.

A couple of possible ways to speed this up:
* Return large packets (512 bytes?) to reduce the number of round trips.
* `JSONDescriptorEndpoint::handle` could cache the schema in memory instead of doing JSON serialization each request.
* Clients can cache the JSON schema and attempt to connect using a precomputed CRC. If the server fails to respond (or modify the server to respond with an error code instead of silence), delete the cache and fall back to the full download.

We're using a variation of the third option where the CRC is specified in a config file, so the JSON schema is not downloaded each time you connect.

samuelsadok, 2020-06-03:
The `devel` and `rc-v0.5.0` branches now contain @adammunich's JSON caching, so when v0.5.0 is released, users should see almost instantaneous connect times in most situations.

Beyond that, as @jhurliman points out, each JSON requests takes quite long because the JSON is rebuilt temporarily for every request. It turns out this is what mainly drives the connect time. This is fixed as a side-effect of #418. With that, I'm seeing connect times of <2s even without caching. That'll probably make it into v0.5.1.

closed 2020-08-29 by madcowswe