odriverobotics/ODrive #359: reserved identifier violation

elfring, 2019-06-21:
I would like to point out that identifiers like “[`__ODRIVE_MAIN_H`](https://github.com/madcowswe/ODrive/blob/901311b582cc6904c4ea76322a8df33d5b961ca6/Firmware/MotorControl/odrive_main.h#L1 "Update candidate")” and “[`__AXIS_HPP`](https://github.com/madcowswe/ODrive/blob/fe0486aa8d246aae8fdc285a4fcaeb38f6fafb54/Firmware/MotorControl/axis.hpp#L1 "Another update candidate")” [do eventually not fit](https://www.securecoding.cert.org/confluence/display/cplusplus/DCL51-CPP.+Do+not+declare+or+define+a+reserved+identifier#DCL51-CPP.Donotdeclareordefineareservedidentifier-NoncompliantCodeExample%28HeaderGuard%29 "Do not declare an identifier which is reserved for the compiler implementation.") to the expected naming convention of the C++ language standard.
Would you like to adjust your selection for unique names?

Wetmelon, 2020-04-26:
If I do a search for `#define __` in the codebase, I get 4082 results lol.  All of the STM core breaks the rules, so I think it's a bit of a lost cause... but we can certainly go through and make the ODrive code compliant

elfring, 2020-04-26:
How do you think about to avoid that this software [depends on undefined behaviour](https://www.securecoding.cert.org/confluence/display/c/CC.+Undefined+Behavior#CC.UndefinedBehavior-ub_106 "Avoiding undefined behaviour?")?

Wetmelon, 2020-11-10:
Oh, you were referring to #106.  Yes, using reserved identifiers is undefined behaviour by the ISO standard, but it's very low on our priority list.  If you feel very strongly about it, please submit a pull request with the corrections

closed 2020-11-10 by Wetmelon