odriverobotics/ODrive #361: find_idx_on_lockin_only should allow all locking states to find the index, not only the steady speed state.

je310, 2019-06-21:
From what I can gather find_idx_on_lockin_only is used to not allow the index pin to be triggered on things like STATE_IDLE or during closed loop control. Though it also excludes LOCKIN_STATE_RAMP and LOCKIN_STATE_ACCELERATE which means that often the motor will need to spin past the index location. I think it should include all LOCKIN states, and exclude all non-lockin states.

madcowswe, 2019-07-22:
The original intention with this setting was to be able to use the rising edge of a hall sensor as index. In this application it had to be completely certain the motor had a positive velocity (with margin) to make sure the rising edge = clockwise edge (or similarly for whatever polarity applies).

You could either have a very short acceleration ramp, or we can make a sparate setting like "allow_idx_find_always", and rename the current setting to "find_idx_on_lockin_full_speed_only" or similar.

Wetmelon, 2021-09-13:
Added this feature some time ago...

closed 2021-09-13 by Wetmelon