odriverobotics/ODrive #374: Set encoder pre-calibrated is not possible without index pin

tboschman, 2019-09-13:
When trying to save the encoder config the pre_calibrated flag always goes back to False.

in encoder.cpp:
Is it possible that this `!index_found_` check should be disabled when no index is used?
```
void Encoder::check_pre_calibrated() {
    if (!is_ready_)
        config_.pre_calibrated = false;
    if (config_.mode == MODE_INCREMENTAL && !index_found_)
        config_.pre_calibrated = false;
}
```

madcowswe, 2019-09-16:
The idea is that you are not allowed to save an encoder index-based calibration if you haven't found the index. Although I could see that this is an issue if you are restoring the config from a backup file.
Is this what you are doing?

tboschman, 2019-09-17:
What we want is to set the drive in closed_loop regulation after power-up without going through the encoder offset calibration.
Our encoder does not have an index pin.

madcowswe, 2019-09-18:
Okay, that is not supported. You need an index pin to start closed loop
without offset calibration when using an incremental encoder. This code is
intended to block such an attempt, as it would be dangerous: without
correct offset information the motor can spin out of control.

closed 2020-02-02 by Wetmelon