odriverobotics/ODrive #379: odrivetool troubleshooting error codes

zwn, 2019-10-04:
It would be helpful if odrivetool shell would somehow translate error codes to error strings. For example I forgot to adjust `odrv0.axis0.motor.config.calibration_current` to some larger value and when I tried to spin the motor, it suddenly stopped. I checked for errors and found
```
In [49]: odrv0.axis0.error                                                                                                                                                       
Out[49]: 32

In [50]: odrv0.axis0.motor.error                                                                                                                                                 
Out[50]: 16

In [51]: odrv0.axis0.encoder.error                                                                                                                                               
Out[51]: 0
```
Then I needed to get back online to find out what that means. The axis error is
```c++
ERROR_MOTOR_DISARMED = 0x20, //<! the motor was unexpectedly disarmed
```
and the motor error is
```c++
ERROR_CONTROL_DEADLINE_MISSED = 0x0010,
```

Actually it would be really helpful that when odrive gets to error state, it would just automatically said what is wrong, without the user involved. Currently (correct me if I wrong) the steps when something unexpected happens are
1. check `axisX.error`, if not zero, goto source code, convert the actual decimal value to hex and try to find it there
2. once you know what the top level error is, goto proper submodule and check error there, like `axisX.motor.error` and goto source code, convert the value to hex and try to match it
3. etc.

Could the odrive tool just say something along the lines of
```
odrv0.axis0.error: motor was unexpectedly disarmed
odrv0.axis0.motor.error: control deadline missed (not enough current?)
```
by itself?

zwn, 2019-10-04:
Here some related info from the forums
* Oskar explains how the error codes are related top to bottom: https://discourse.odriverobotics.com/t/troubleshooting-calibration-errors/916/4?u=zwn
* Oskar asking a user to readout errors after motor unexpectedly stops: https://discourse.odriverobotics.com/t/troubleshooting-calibration-errors/916/10?u=zwn

zwn, 2019-10-07:
Some more info:

> You have encoder error ERROR_ILLEGAL_HALL_STATE = 0x10. This means that all hall signals are 0 or all are 1: this is not allowed and means something is wrong with the hall feedback. Please check your wiring and if it looks good, please double check behaviour with an oscilloscope or logic analyzer.

https://discourse.odriverobotics.com/t/encoder-error-error-illegal-hall-state/1047/2?u=zwn

I imagine this is what odrivetool should say when `ERROR_ILLEGAL_HALL_STATE` is encountered - maybe also adding something about adding the 22nF capacitors :smile:

JB-VA, 2019-12-04:
Use `dump_errors(axis, True)` where axis if your axis number, and True tells it to clear the error after displaying a human readable error codes like what you are looking for.

closed 2020-08-29 by madcowswe