odriverobotics/ODrive #381: Documentation for python API

zwn, 2019-10-09:
It would be great to have a documentation to the package available at https://pypi.org/project/odrive/. There are some examples throughout the https://docs.odriverobotics.com/ but actually having a list of all packages, functions and classes with one sentence describing each, would be best.

samuelsadok, 2020-05-20:
I assume you're referring not to the function/classes of the python package but specifically to the functions and properties exported by the ODrive itself, such as `odrv0.axis0.requested_state`?
If that's the case then that's definitely desirable and on the radar.

samuelsadok, 2020-08-31:
An initial version of a HTML API reference was merged to `devel` with #424 and will go live on https://docs.odriverobotics.com when firmware v0.5.1 is released.

The source of the documentation is [odrive-interface.yaml](https://github.com/madcowswe/ODrive/blob/devel/Firmware/odrive-interface.yaml). Feel free to enhance the documentation in that file.

Here's a preview of what it looks like when compiled to HTML:

[image]

closed 2020-08-31 by samuelsadok