odriverobotics/ODrive #394: Bug connecting to ODrive with two versions of odrivetool

LarsRhijns, 2019-12-12:
I have been experiencing a bug when trying to connect to the odrive while there are two versions of odrive installed (one via `$ pip install odrive` and one via `$ pip3 install odrive`). I don't know whether this is actually a bug or just something stupid I did without noticing, but I though I would elaborate it here so it can be prevented or documented somewhere so other won't have to struggle as well. 

When I have odrive (v0.4.11) installed via pip and pip3, I was able to run odrivetool, but I got the following warnings: 

`$ [path_to_python3.6]/site-packages/fibre/shell.py:104: RuntimeWarning: coroutine 'InteractiveShell.run_code' was never awaited
  console.runcode('import sys')
$ [path_to_python3.6]/site-packages/fibre/shell.py:105: RuntimeWarning: coroutine 'InteractiveShell.run_code' was never awaited
  console.runcode('superexcepthook = sys.excepthook')
$ [path_to_python3.6]/site-packages/fibre/shell.py:106: RuntimeWarning: coroutine 'InteractiveShell.run_code' was never awaited
  console.runcode('def newexcepthook(ex_class,ex,trace):\n'
$ [path_to_python3.6]/site-packages/fibre/shell.py:109: RuntimeWarning: coroutine 'InteractiveShell.run_code' was never awaited
  console.runcode('sys.excepthook=newexcepthook')
`

As a result the ODrive board (v3.6, 56V) wouldn't connect to the utility tool. I have tried with two seperate ODrive boards, same result. 

When uninstalling either the pip or pip3 version of the odrive utility tool and running `odrivetool shell` again with the ODrive board connected, it works fine.

Again I don't know if this is really a bug/issue, but maybe this info is useful.

bsamadi, 2019-12-16:
I have the same problem. I have installed odrive using only pip3 on Windows and Ubuntu 18.04.

samuelsadok, 2020-05-26:
@LarsRhijns the side-by-side installation in python2 and python3 shouldn't be an issue per se, it just means that it's somewhat arbitrary which one is used when you just run `odrivetool`.

The warnings you're seeing are a known issue, they are harmless and fixed on the `devel` branch (will make it to the next release). @bsamadi you probably meant these warnings when you said you have the same problem?

Now to your actual connect problem: When you install odrive without sudo, then the udev rules won't be set up correctly and odrivetool won't connect as a result (because it can't access the USB). We should probably add a helpful message for this case. Is it possible that on your second attempt (the one that worked) you did it with sudo?

samuelsadok, 2020-08-31:
Closing this due to inactivity, but feel free to reopen if you still have issues with `odrivetool` on the `devel` branch.

closed 2020-08-31 by samuelsadok