odriverobotics/ODrive #397: Possible Race Condition at startup using stand-alone operation with PWM

emdash, 2020-01-29:
What happens: between 10% and 25% of the time, motor is driven full reverse shortly after power-on or soft reset, despite the PWM signal commanding a slow speed, or zero speed.

I'm not sure if this is caused by the PWM mapping or a bug in the startup sequence. When it does reproduce, it seems to happen immediately -- hard to tell if the index search is running or not.

We are running vanilla 0.4.11 firmware on an ODrive 3.6. I am attaching both a dump of the current config, as well as a python script which, when run from within `odrivetool` will replicate the config.

Steps to reproduce:
* Configure for stand-alone operation, velocity control, using a PWM input.
  * Our mapping is bi-directional, so a zero velocity is in the middle of the duty-cycle range.
* Connect an appropriate PWM signal.
* Power-cycle or reboot the odrive. Occasionally, the motor will run full reverse shortly after boot.

Config.py:
```
# This script is designed to run from within the odrivetool shell, via
# IPython's "%load" magic. It is NOT a plain python script.
import time

print("Erasing previous configuration")
odrv0.erase_configuration()
try:
    odrv0.reboot()
except fibre.ChannelBrokenException:
    time.sleep(20)

print("Setting initial config values")
# Cache the axis value.
axis_name = 'axis0'
axis = getattr(odrv0, axis_name)

# Defined by our motor confugration.
axis.motor.config.current_lim = 60
enc_counts_per_rotation = 4096
axis.encoder.config.cpr = enc_counts_per_rotation
vel_limit_rpm = 2500
vel_limit_cpr = vel_limit_rpm * enc_counts_per_rotation / 60
axis.controller.config.vel_limit = vel_limit_cpr

# I raise this up a bit depending on the motor, but 10 A seems fine
axis.motor.config.calibration_current = 10

# This is 4 for the nanotech motors we use
axis.motor.config.pole_pairs = 4

pwm = odrv0.config.gpio3_pwm_mapping
pwm.min = -vel_limit_cpr
pwm.max = vel_limit_cpr
pwm.endpoint = axis.controller._remote_attributes['vel_setpoint']

# There is a known bug that happens after saving configuration,
# rebooting immediately after fixes this issue.
odrv0.save_configuration()
try:
    odrv0.reboot()
except fibre.ChannelBrokenException:
    # This delay could be shorter, but we need to stopwatch it.
    time.sleep(20)

# Re-cache our axis, since we have lost the connection.
axis = getattr(odrv0, axis_name)

print("Running calibration sequence")
# Run the calibration sequence
axis.requested_state = AXIS_STATE_FULL_CALIBRATION_SEQUENCE
# There is no other way to tell when the calibration sequence has
# finished.
time.sleep(20)

print("Configure index")
axis.encoder.config.use_index = True
[... 174 lines cut ...]
            "config": {
                "observer_gain": 1000.0,
                "pll_bandwidth": 1000.0,
                "pm_flux_linkage": 0.0015800000401213765
            }
        },
        "trap_traj": {
            "config": {
                "A_per_css": 0.0,
                "accel_limit": 5000.0,
                "decel_limit": 5000.0,
                "vel_limit": 20000.0
            }
        }
    },
    "config": {
        "brake_resistance": 2.0,
        "dc_bus_overvoltage_trip_level": 59.92000198364258,
        "dc_bus_undervoltage_trip_level": 8.0,
        "enable_ascii_protocol_on_usb": true,
        "enable_i2c_instead_of_can": false,
        "enable_uart": true
    }
}
```

Wetmelon, 2020-02-02:
This is using your 0-100% PWM speed command interface, right?  What thread is that running on?

emdash, 2020-02-04:
No, this is all using vanilla ODrive FW. I tried on 0.4.11, and on Razor's
edge. The specifics of how the bug gets triggered are different with each
revision, but the failure mode is the same.

Also, it's not (yet) my 0-100% PWM. I'm just interested in this
functionality. I have reviewed the changes, but don't yet undertand them
100%. There are some changes to the CubeMX code that the core devs would
really need to vet, but I think they might be related to what's going on
here.

emdash, 2020-02-06:
I tried removing the PWM mapping from the configuration, using UART and odrivetool. I can reproduce a similar failure mode by commanding a velocity setpoint of `30719` (in counts). This is about 450 RPM, and a small fraction of the 2500 rpm limit (`170666` counts).

So I don't think this issue is what I thought it was at first. It seems like the error is on the output side, not the input side, or anything to do with calibration.

emdash, 2020-02-06:
This issue doesn't reproduce every time. I am not sure exactly what causes it, but my guess is it has to do with the exact position of the rotor at startup. I power-cycled the same setup as mentioned previously -- without PWM. After about 6 or so attempts, the following happened:

- rotor briefly spun up, then shut off
- `dump_errors(odrv0)` reported the following:

[image]

emdash, 2020-02-06:
Tried removing `startup_closed_loop_control` from the saved config, still observed this issue when commanding the motor to run -- required power cycling about 6 times. Commanded a slow speed via UART ("v 0 30719"), motor spins full reverse. Then apparently goes into an error state -- cannot connect via odrivetool.

emdash, 2020-02-06:
While it's in this broken state, ran the following from within `odrivetool`:

```
In [17]: odrv0.axis0.requested_state = AXIS_STATE_CLOSED_LOOP_CONTROL; odrv0.axis0.controller.vel_setpoint = 30719; print(odrv0.axis0); print(odrv0.axis0.controller); print(odrv0.axis0.motor); time.sleep(1); odrv0.axis0.requested_sta
    ...: te = AXIS_STATE_IDLE;                                                                                                                                                                                                           
error = 0x0000 (int)
sensorless_estimator:
  error = 0x0000 (int)
  pll_pos = -2.8082170486450195 (float)
  config: ...
  phase = -2.8103199005126953 (float)
  vel_estimate = 7.131326198577881 (float)
watchdog_feed()
loop_counter = 8778744 (int)
config:
  sensorless_ramp: ...
  startup_encoder_index_search = True (bool)
  startup_sensorless_control = False (bool)
  dir_gpio_pin = 2 (int)
  calibration_lockin: ...
  startup_motor_calibration = False (bool)
  general_lockin: ...
  startup_closed_loop_control = False (bool)
  enable_step_dir = False (bool)
  counts_per_step = 2.0 (float)
  step_gpio_pin = 1 (int)
  watchdog_timeout = 0.0 (float)
  startup_encoder_offset_calibration = False (bool)
encoder:
  calib_scan_response = 0.0 (float)
  error = 0x0000 (int)
  hall_state = 3 (int)
  phase = 1.8222832679748535 (float)
  pos_estimate = -2380398.25 (float)
  config: ...
  vel_estimate = -17625.001953125 (float)
  is_ready = True (bool)
  pos_cpr = 3450.921142578125 (float)
  count_in_cpr = 3435 (int)
  shadow_count = -2380444 (int)
  set_linear_count(count: int)
  interpolation = 1.0 (float)
  index_found = True (bool)
trap_traj:
  config: ...
current_state = 8 (int)
step_dir_active = False (bool)
controller:
  error = 0x0000 (int)
  vel_setpoint = 30719.0 (float)
  current_setpoint = 0.0 (float)
  pos_setpoint = -2380334.25 (float)
  vel_integrator_current = 0.4949754774570465 (float)
  set_current_setpoint(current_setpoint: float)
  set_pos_setpoint(pos_setpoint: float, vel_feed_forward: float, current_feed_forward: float)
  start_anticogging_calibration()
  config: ...
  vel_ramp_target = 0.0 (float)
  move_to_pos(pos_setpoint: float)
  set_vel_setpoint(vel_setpoint: float, current_feed_forward: float)
  move_incremental(displacement: float, from_goal_point: bool)
  vel_ramp_enable = False (bool)
motor:
  error = 0x0000 (int)
  is_calibrated = True (bool)
  phase_current_rev_gain = 0.02500000037252903 (float)
  DC_calib_phB = -0.4713646173477173 (float)
  armed_state = 3 (int)
  current_meas_phC = 31.31407356262207 (float)
  config: ...
  get_inverter_temp()
  DC_calib_phC = -1.140409231185913 (float)
  current_control: ...
  timing_log: ...
  gate_driver: ...
  thermal_current_lim = 248.3859405517578 (float)
  current_meas_phB = -37.43398666381836 (float)
lockin_state = 0 (int)
requested_state = 0 (int)
error = 0x0000 (int)
vel_setpoint = 30719.0 (float)
current_setpoint = 0.0 (float)
pos_setpoint = -2380334.25 (float)
vel_integrator_current = 1.008988618850708 (float)
set_current_setpoint(current_setpoint: float)
set_pos_setpoint(pos_setpoint: float, vel_feed_forward: float, current_feed_forward: float)
start_anticogging_calibration()
config:
  vel_gain = 0.0005000000237487257 (float)
  vel_ramp_rate = 10000.0 (float)
  vel_limit_tolerance = 1.2000000476837158 (float)
  setpoints_in_cpr = False (bool)
  vel_limit = 170666.671875 (float)
  pos_gain = 20.0 (float)
  control_mode = 2 (int)
  vel_integrator_gain = 0.0010000000474974513 (float)
vel_ramp_target = 0.0 (float)
move_to_pos(pos_setpoint: float)
set_vel_setpoint(vel_setpoint: float, current_feed_forward: float)
move_incremental(displacement: float, from_goal_point: bool)
vel_ramp_enable = False (bool)
error = 0x0000 (int)
is_calibrated = True (bool)
phase_current_rev_gain = 0.02500000037252903 (float)
DC_calib_phB = -0.47626104950904846 (float)
armed_state = 3 (int)
current_meas_phC = 18.42047119140625 (float)
config:
  current_control_bandwidth = 1000.0 (float)
  motor_type = 0 (int)
  current_lim = 60.0 (float)
  pre_calibrated = True (bool)
  requested_current_range = 60.0 (float)
  direction = -1 (int)
  phase_inductance = 0.0001915805332828313 (float)
  pole_pairs = 4 (int)
  current_lim_tolerance = 1.25 (float)
  inverter_temp_limit_lower = 100.0 (float)
  phase_resistance = 0.06781734526157379 (float)
  resistance_calib_max_voltage = 2.0 (float)
  calibration_current = 10.0 (float)
  inverter_temp_limit_upper = 120.0 (float)
get_inverter_temp()
DC_calib_phC = -1.1349217891693115 (float)
current_control:
  Ibus = 4.005885601043701 (float)
  max_allowed_current = 60.75 (float)
  overcurrent_trip_level = 67.5 (float)
  i_gain = 67.81734466552734 (float)
  Iq_measured = -22.016376495361328 (float)
  v_current_control_integral_d = 8.44771671295166 (float)
  Id_measured = -21.359893798828125 (float)
  I_measured_report_filter_k = 1.0 (float)
  final_v_beta = 6.131686210632324 (float)
  p_gain = 0.19158053398132324 (float)
  final_v_alpha = -9.904891967773438 (float)
  v_current_control_integral_q = -10.944363594055176 (float)
  Iq_setpoint = -60.0 (float)
timing_log:
  TIMING_LOG_FOC_CURRENT = 7542 (int)
  TIMING_LOG_GENERAL = 0 (int)
  TIMING_LOG_FOC_VOLTAGE = 0 (int)
  TIMING_LOG_ENC_CALIB = 0 (int)
  TIMING_LOG_MEAS_L = 0 (int)
  TIMING_LOG_ADC_CB_I = 2618 (int)
  TIMING_LOG_ADC_CB_DC = 12862 (int)
  TIMING_LOG_IDX_SEARCH = 0 (int)
  TIMING_LOG_MEAS_R = 0 (int)
gate_driver:
  drv_fault = 0 (int)
thermal_current_lim = 248.56224060058594 (float)
current_meas_phB = 19.648773193359375 (float)
```

emdash, 2020-02-07:
Disabled `axis0.encoder.config.pre_calibrated' and `axis0.motor.config.pre_calibrated` and saved config. Power cycled, and ran the full calibration sequence.

odrv0.axis0.encoder now looks more like:

```
In [7]: odrv0.axis0.encoder                                                  
Out[7]: 
pos_cpr = 360.5864562988281 (float)
vel_estimate = 102375.0078125 (float)
set_linear_count(count: int)
count_in_cpr = 487 (int)
phase = -2.914822578430176 (float)
calib_scan_response = 8119.0 (float)
shadow_count = 1962599 (int)
hall_state = 2 (int)
interpolation = 0.0 (float)
index_found = True (bool)
config:
  calib_scan_distance = 50.26548385620117 (float)
  bandwidth = 1000.0 (float)
  use_index = True (bool)
  ignore_illegal_hall_state = False (bool)
  cpr = 4096 (int)
  pre_calibrated = False (bool)
  calib_scan_omega = 12.566370964050293 (float)
  offset_float = 0.041999995708465576 (float)
  zero_count_on_find_idx = True (bool)
  enable_phase_interpolation = True (bool)
  offset = -3083 (int)
  idx_search_unidirectional = False (bool)
  calib_range = 0.019999999552965164 (float)
  mode = 0 (int)
  find_idx_on_lockin_only = False (bool)
is_ready = True (bool)
error = 0x0000 (int)
pos_estimate = 1963752.125 (float)
```

emdash, 2020-02-22:
I have instrumented the code on a fork, and was able to get some possibly useful data:

This is an excerpt from a "good" run. The values being logged are `vel_des`, a value derived from the velocity setpoint, `vel_estimate`, the current encoder velocity, and the `Iq`, which is effectively the current setpoint.
```
debug: dbd8    27049 vel_des: 4096.000000
debug: dbd8    27049 vel_estimate: 3625.000488
debug: dbd8    27049 Iq: 0.864743
debug: dbd8    27149 vel_des: 4096.000000
debug: dbd8    27149 vel_estimate: 3500.000488
debug: dbd8    27149 Iq: 0.926767
debug: dbd8    27249 vel_des: 4096.000000
debug: dbd8    27249 vel_estimate: 4750.000488
debug: dbd8    27249 Iq: 0.317443
debug: dbd8    27349 vel_des: 4096.000000
```

This is from an earlier run, where I didn't have the current setpoint output. As soon a

```
debug: dbd8    15179 vel_estimate: 0.000000
debug: dbd8    15279 vel_setpoint: 0.000000
debug: dbd8    15279 vel_des: 0.000000
debug: dbd8    15279 vel_estimate: 0.000000
debug: dbd8    15379 vel_setpoint: 4096.000000
debug: dbd8    15379 vel_des: 4096.000000
debug: dbd8    15379 vel_estimate: -124750.007812
debug: dbd8    15479 vel_setpoint: 4096.000000
debug: dbd8    15479 vel_des: 4096.000000
debug: dbd8    15479 vel_estimate: -154000.015625
debug: dbd8    15579 vel_setpoint: 4096.000000
```
ERROR_OVERSPEED was not triggered on the run above because our vel limit is actually > 170000. Is there a hard RPM limit?

Here's some data from a bad run where ERROR_OVERSPEED was triggered. One thing I find telling is that `vel_estimate` is 0, indicating that sometimes a problems with the encoder logic.

```
debug: dbd8    18190 run_closed_loop_control_loop(): enter
debug: dbd8    18190 control_mode: 2
debug: dbd8    18190 pos_setpoint: 300.234375
debug: dbd8    18190 vel_setpoint: 0.000000
debug: dbd8    18190 current_setpoint: 0.000000
debug: dbd8    18190 run_control_loop()
debug: dbd8    18261 vel_setpoint: 0.000000
debug: dbd8    18261 vel_des: 0.000000
debug: dbd8    18261 vel_estimate: 0.000000
debug: dbd8    18261 Iq: -0.002516
debug: dbd8    18361 vel_setpoint: 0.000000
debug: dbd8    18361 vel_des: 0.000000
debug: dbd8    18361 vel_estimate: 0.000000
debug: dbd8    18361 Iq: -0.002516
debug: dbd8    18461 vel_setpoint: 0.000000
debug: dbd8    18461 vel_des: 0.000000
debug: dbd8    18461 vel_estimate: 0.000000
debug: dbd8    18461 Iq: -0.002516
debug: dbd8    18561 vel_setpoint: 0.000000
debug: dbd8    18561 vel_des: 0.000000
debug: dbd8    18561 vel_estimate: 0.000000
debug: dbd8    18561 Iq: -0.002516
debug: dbd8    18661 vel_setpoint: 0.000000
debug: dbd8    18661 vel_des: 0.000000
debug: dbd8    18661 vel_estimate: 0.000000
debug: dbd8    18661 Iq: -0.002516
```

emdash, 2020-03-04:
This turned out to be a hardware issue and not a firmware bug -- noise on the encoder inputs. Adding two 1nF caps to the index signal, as per the forum thread, eliminated the problem in our testing.  We are exploring an additional hardware solution as well that completely eliminates noise on the the scope trace for all encoder inputs.

As an additional safeguard, we may try to implement "setpoint relative" over-speed detection, as the existing overs-speed detection only considers the the absolute limit. This does seem tricky to get right, however.

In the course of investigating this bug, we discovered a reliable way to reproduce it with our setup that involves setting the rotor to a specific position prior to performing an index search.

Wetmelon, 2020-04-26:
According to OP, issue ended up being hardware related

closed 2020-04-26 by Wetmelon