odriverobotics/ODrive #404: get_inverter_temp through ASCII no response

Peter-haotian-cui, 2020-04-24:
Hi,

I use "r axis0.motor.get_inverter_temp" command for getting axis0 inverter temperature, but there is no response. Is it my command wrong? any suggestions? 
vbus_voltage and other commands are usable.

Thanks

Wetmelon, 2020-04-26:
See forum post: https://discourse.odriverobotics.com/t/driving-two-motors-parallel/4550/3?u=wetmelon

samuelsadok, 2020-05-20:
The `r` command of the ASCII protocol only supports reading properties whereas `axis0.motor.get_inverter_temp()` is (was) a function. With @Wetmelon's changes it's exposed as a property now (merged to the `devel` branch). That means you can read the temperature with `r axis0.motor.inverter_temp`.

@Wetmelon the forum post seems unrelated, probably mixed up a link there.

Wetmelon, 2020-07-31:
Workaround implemented, closing for now

closed 2020-07-31 by Wetmelon